AlbrechtL/welle.io #337: Black background

jpuigs, 2019-01-08:
Is it possible to go back to dark gray background, as it was in version 1 ?
The white background on version 2 is annoying.

lars18th, 2019-01-09:
Hi,

I suggest to support multiple predefined color backgrounds: white, dark, gray, etc.

Personally, I don't like much the need look & feel. But, any improvement is welcome.

Thank you for this great project!

AlbrechtL, 2019-01-10:
The styling is based on https://doc.qt.io/qt-5/qtquickcontrols2-styles.html. So adding more styles are technical possible. For now this isn't a high priority task.

If someone sends a pull request we can discuss about it.

AlbrechtL, 2019-03-03:
I' closing this issue for now because it has a low priority. But we can use this issue to discuss it more if required.

closed 2019-03-03 by AlbrechtL