AlbrechtL/welle.io #338: Raspi documentation is outdated

mattes-bru, 2019-01-08:
I recently set up a raspi to run welle-io and found that the documentation needs an update:

[ ] the new dependencies of 2.0 are missing
[ ] I needed to do a shadow build of Qt with some other configure options to make Qt 5.12.0 compile
[ ] the sample application is misleading (it's only dependency is qtbase, whereas welle-io needs more Qt modules)
[ ] remote deployment / debugging through Qt Creator is missing
[ ] the known issues are not really an issue rather a misconfiguration welle-io is not using the window manager so you could disable the window-manager and start welle-io with systemd 

I started to work on this in my forked repo. I'll do a pull request as soon as I am done.

AlbrechtL, 2019-01-10:
Thanks for reporting. Feel free to update it and send a pull request.

mattes-bru, 2019-01-11:
I updated the readme in my fork. This is still work in progress but should end in welle-io 2.0 running on the pi. I wrote this by checking my bash history but without actually doing the steps. So any comments, tests are helpful before sending a pull request.

FartherOut, 2019-01-13:
Very nice.
My guide is woefully outdated for v2.0 stuff and never packages/Qt versions, and I haven't really been able to put in the work in updating it properly.
Also, I am probably not the best person for this type of documentation, as I can barely understand anything Linux at all. :)

By sample application I assume you mean the qopenglwidget example which is included in Qt?
I included it in the guide to simply check if one can get Qt to even run properly on the Pi, not that it means everything is in place to make welle.io work.
Perhaps this should be highlighted better i the guide, I don't know.

mattes-bru, 2019-01-13:
@FartherOut your right I am referring to the QOpenGLWidget example. My problem was that I had only qtbase installed which is sufficient for this example project but not for welle-io. But now, a couple days later I think a quick test makes sense in this guide. 

So right now I need to do the documentation for the systemd autostart. The scripts are already working: welle-io is starting after boot either in eglfs or headless with webgl (one nice thing about Qt 5.12).

I also want to add a shortcut to the application menu, so that welle-io can be started inside the X-Server.

Has anybody tried my instructions so far? I am really interested in feedback.

AlbrechtL, 2019-03-03:
How is the progress here?

mattes-bru, 2019-03-04:
There is one thing I wanted to do, but I didn't find the time for it. I wanted to add entry to the application menu, which starts welle.io in desktop mode. Beside from that, it is done.

AlbrechtL, 2019-03-06:
I you feel to create a pull request just go for it.

AlbrechtL, 2019-08-10:
Any news here?

mattes-bru, 2019-08-12:
I added Autostart through systemd (in embedded mode) and Applications Menu entry. But I didn't find the time to update the documentation for this.

wshadow, 2019-12-17:
> Has anybody tried my instructions so far? I am really interested in feedback. There are two main things I am unsure about:
> 
>     * Step 16: some of the links must have already existed
> 
>     * wether it is good to include soapysdr in the build. I had some crashes when the input is set to automatic.

@mattes-bru I'm currently following your instructions. I'm using a fresh install of raspbian buster with desktop (rel.date: 2019-09-26). In your step 18 (Fix the EGL/GLES libraries.) some filenames/paths seem to have changed. /usr/lib/arm-linux-gnueabihf/libEGL.so.1.0.0 and /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0.0 didn't exist on my system.

Running qopenglwidget also doesn't quite work:

```
$ /home/pi/qopenglwidget 
qt.qpa.plugin: Could not find the Qt platform plugin "eglfs" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, minimal, offscreen, vnc, wayland, webgl, xcb.
```

It does however run, if I pass the parameter -platform linuxfb:
```
$ /home/pi/qopenglwidget -platform linuxfb
QOpenGLWidget is not supported on this platform.
This plugin does not support createPlatformOpenGLContext!
This plugin does not support createPlatformOpenGLContext!
QOpenGLWidget: Failed to create context
This plugin does not support createPlatformOpenGLContext!
QOpenGLWidget: Failed to create context
This plugin does not support createPlatformOpenGLContext!
QOpenGLWidget: Failed to create context
```

Any suggestions on what's wrong here?

I also had some missing packets on my host to build qt, but that seems to be solved. I'll try to find the names for the missing packages in my history and will add them here later.

I'll let you know if I find anything else.

Ps.: You could have made finding the instructions easier by mentioning the branch name raspi somewhere here ;-)

mattes-bru, 2019-12-20:
@wshadow This is something I also stumbled across the last time I wanted to finish this: The eglfs platform is not build during the Qt build process. Right now I don't have a idea what's wrong here. I was thinking about the update from Qt 5.11 to 5.12. 

But depending on how you want to use welle-io the is not a big issue. You only need the eglfs platform when you want to boot directly into welle-io on your pi. 

If you want to use welle-io as an application on your pi desktop, try the xcb platform. You need to start qopenglwidget from the raspi desktop in this case. 

The linuxfb platform directly writes to the framebuffer without using the window manager and does not have OpenGL support as far as I know. So the qopenglwidget does not work with this platform but it might work for welle-io itself (the software showed up here but I didn't tested further), if you really want to use it. 

Did welle-io get welle-io compiled / deployed? if so, you should find a welle-io shortcut in the raspi desktop application menu.

wshadow, 2019-12-22:
Well, I guess I'll not use eglfs for now and try to use xcb.

I still have some problems with qtcreator. Also I think I messed up my qt versions somewhere along the way. Probably because I was following about 4 different tutorials to get eglfs to work. I'll start from scratch after Christmas and report back once I get it to work or find a new problem.

alexmyczko, 2020-03-03:
meanwhile rapsbian testing has the debian package https://repology.org/project/welle.io/versions

FartherOut, 2020-07-01:
I tried to follow this guide with Lubuntu 18.04, linaro 7.5.0, qt 5.15.0.
I had to manually install python(2.7) and libxkbcommon-x11-0.
However, it seems that one suddenly need to have a QT account in order to use the QT online installer, which wasn't the case before.

So I'm currently stuck not being able to use QT because I can't be bothered to make an account when I didn't have to before.
Some people have suggested using the offline installer and disconnect from the web for the "skip" button to appear once again.
But I have no idea how the offline installer even works. (Just get a ton of warnings and OpenGL errors)

andimik, 2020-07-02:
I even found a fake account. Please continue.

FartherOut, 2020-07-03:
Used some account and manage to compile it.
Looks like the manual is working correctly for the time being.

Don't know about the auto deploying though. It sent the program over to the pi but it didn't start it, maybe I did it wrong.
Anyways I tend to do stuff like this manually.

Three minor issues.
1, the mute button icon is invisible when you mute, which kinda makes it hard to unmute unless you know the button is there.
2, MOT slideshow now shows two debug lines "image://sls/CategoryID/CategoryTitle/SlideID/ContentName" and "image://sls/empty" (which changes to whatever image name/number it receives from the station)
3, some text is not translated, such as MOT slideshow "Save all images" option. And the entire "Style settings" section.

Material theme is transparent which makes it almost impossible to select anything in menus (including changing the theme back into something else :) )

Otherwise, cool.

basicmaster, 2020-07-04:
> However, it seems that one suddenly need to have a QT account in order to use the QT online installer, which wasn't the case before.
>
> So I'm currently stuck not being able to use QT because I can't [...]

Indeed Qt unfortunately did a (repeated!) try to force people to register in order to install the framework. Furthermore, more recent offline installers will only be available if you pay. But it indeed works to use an older version and to disable the internet connection during the installation.

AlbrechtL, 2020-07-07:
@FartherOut Congratulations! Feel free to update the RPi manual by sending a PR.

For your 3 issues open 3 new github issues with screenshots please.

AlbrechtL, 2022-12-16:
Closed because this issue is outdated. welle.io is included inside Raspberry OS.

closed 2022-12-16 by AlbrechtL