AlbrechtL/welle.io #341: [welle-gui] Translations outdated

AlbrechtL, 2019-01-30:
Due internal reorganizing of the code the most translations are broken.

The fix this translators can simply run `lupdate` again (updating the TS files) and then run "Qt Linguist" to create the QM files.

By default most of the translations should work again but I recommend to add new translation as well.

The full process to add new translations is documented here: https://github.com/AlbrechtL/welle.io/wiki/Translating-welle.io

AlbrechtL, 2019-01-30:
@meeuw, @BRychlik, @olilesp, @andimik Your are really welcome to update your translation by sending a pull request.

FartherOut, 2019-08-13:
Hmm yeah. I finally updated my translations but I can't seem to get lrelease to make a qm file. In QT Creator, all I get is:

WARNING: Project ERROR: Unknown module(s) in QT: androidextras
Segmentation fault (core dumped)
"/opt/Qt/5.13.0/gcc_64/bin/lrelease" finished

AlbrechtL, 2019-08-14:
Did you try to install the `androidextras` package?

FartherOut, 2019-08-14:
No idea what's going on. There are 3 folders called androidextras in my installation but I don't know why it wouldn't work. Anyways, tried commenting out the condition code and still got a segmentation fault error.

tenzap, 2019-08-15:
I updated the french translation. See https://github.com/AlbrechtL/welle.io/issues/388
Not through PR, but file is attached :)

FartherOut, 2019-09-04:
Hey I think I got it by doing a release inside linguist itself, and I managed to compile welle.
However welle.io -L doesn't work anymore so I can't check to see if the translations looks OK. Any suggestions?

welle.io: Unknown option 'L'.

FartherOut, 2019-09-05:
Did the nuclear option and changed the language of my OS so I could check out the new translations. I have now completed and verified nb_NO. It sits in pull request #395

AlbrechtL, 2019-09-06:
> welle.io: Unknown option 'L'.

I removed this option and added a drop down menu inside the settings page.
[image]

AlbrechtL, 2019-12-02:
Closed for now. But everyone is still invited to send pull requests with language updates.

closed 2019-12-02 by AlbrechtL