AlbrechtL/welle.io #344: NooElec R820T SDR & DVB-T no channels found

Nullstr1ng, 2019-02-03:
I have NooElect R820T SDR & DVB-T NESDR Mini USB kit and welle.io did not find any channel. It was scanning alright even if the driver auto detect is enabled or disabled by using RTL-SDR option, but still, no channels found.

I tried using SDR# and set the frequency  to our Digital TV freq signal and the signal was there 

here's the sample.
[image]

Here's the exact model of what I'm using
https://www.youtube.com/watch?v=9lZLKa7QWKY
(video not mine)

andimik, 2019-02-03:
If this is channel 43 (650 MHz) then I wonder why your screenshot starts at 644 MHz and not at 646 MHz as the DVB-T or T2 is 8 MHz, not 10 MHz.

Can you post a screenshot from you VHF DAB Signal.

I fear your dongle has a big offset.

andimik, 2019-02-03:
🇵🇭?

There are no DAB transmissions there. So you seem not to have understood the purpose of welle.io.

https://www.worlddab.org/country-information

Nullstr1ng, 2019-02-03:
Yes, Philippines.

Sorry, I am very new to this one. Do you have any suggestions what software I can use to play Digital TV channels from my device?

andimik, 2019-02-03:
You need to install the original driver as you use the dongle as SDR.

But please ask someone from your country. All countries in Europe use DVB-T or DVB-T2.

The Philippines use the Japanese standard. So we cannot help you at all.

https://en.m.wikipedia.org/wiki/ISDB#ISDB-T

closed 2019-02-12 by AlbrechtL