AlbrechtL/welle.io #346: SoapySDR on home-brew cask build

rabarar, 2019-02-11:
I can't seem to get the SoapySDR  driver to work with the OSX home-brew release. I can see my USRP device via SoapySDRUtil but can't connect with welle-io.

```
$ SoapySDRUtil --make
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device 
[INFO] [UHD] Mac OS; Clang version 10.0.0 (clang-1000.11.45.5); Boost_106800; UHD_3.13.1.0-0-unknown
[INFO] [usrp2_impl.cpp:309] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [usrp2_impl.cpp:346] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [usrp2_impl.cpp:347] [USRP2] Current send frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
[INFO] [usrp2_impl.cpp:527] [USRP2] Detecting internal GPSDO.... 
[INFO] [USRP2] Detecting internal GPSDO.... 
[INFO] [gps_ctrl.cpp:243] [GPS] Found an internal GPSDO: Jackson-Labs, FireFly , Firmware Rev 0.929
[INFO] [GPS] Found an internal GPSDO: Jackson-Labs, FireFly , Firmware Rev 0.929
[INFO] [usrp2_impl.cpp:795] [USRP2] Setting references to the internal GPSDO
[INFO] [USRP2] Setting references to the internal GPSDO
  driver=usrp2
  hardware=N210r4
  fpga_version=11.1
  fw_version=12.4
  mboard_id=N210r4
  mboard_name=
  mboard_serial=F306C0
  rx0_antenna=RX2
  rx0_id=WBX v3, WBX v3 + Simple GDB (0x0057)
  rx0_serial=F35E28
  rx0_subdev_name=WBXv3 RX+GDB
  rx0_subdev_spec=A:0
  tx0_antenna=TX/RX
  tx0_id=WBX v3 (0x0056)
  tx0_serial=F35E28
  tx0_subdev_name=WBXv3 TX+GDB
  tx0_subdev_spec=A:0
```

AlbrechtL, 2019-02-12:
Please post the welle.io console output.

rabarar, 2019-02-13:
> 
> Please post the welle.io console output.

$ welle-io -d soapysdr,driver=uhd,hardware=N210r4
main: Set language "en_US"
main: Error while loading language "en_US" use English "en_GB" instead
Audio: ActiveState
qml: dpiScaleFactor: 1
qrc:/src/gui/QML/RadioView.qml:134:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:142:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:105:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:112:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:20:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:60:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:70:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:20:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:60:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:70:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:20:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:60:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:70:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:20:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:60:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:70:9: QML RowLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:105:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:112:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:105:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/RadioView.qml:112:9: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/src/gui/QML/SettingsPage.qml:73:13: QML ColumnLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qml: Apply settings initially
qml: os: osx
qml: desktopAvailableWidth: 1680
qml: desktopAvailableHeight: 1027
qml: orientation: 2
qml: devicePixelRatio: 2
qml: pixelDensity: 5.07874023375534
Audio: ActiveState
InputFactory: Input device: "soapysdr,driver=uhd,hardware=N210r4"
InputFactory: Unknown device " "soapysdr,driver=uhd,hardware=N210r4" ".
InputFactory: Error while opening device "soapysdr,driver=uhd,hardware=N210r4".
RadioController: AGC on
Audio: Volume 1
Audio: ActiveState
GUI: deleting radioInterface
OFDM-decoder: closing down now
main: Application closed

AlbrechtL, 2019-02-17:
I checked the HomebrewFormula. Unfortunately SoapySDR is not included into the build.
Since I don't have a Mac I can't fix it. 

https://github.com/AlbrechtL/welle.io/blob/4186caa56b48840ad498451b372a709b74ef9759/HomebrewFormula/welle.io.rb#L1-L23

Everyone is welcome to send a pull request with SoapySDR support!

rabarar, 2019-02-18:
Here's the modified Homebrew Formula to build with SoapySDR support

```
class WelleIo < Formula
  desc "DAB/DAB+ Software Radio"
  homepage "https://www.welle.io"
  url "https://github.com/AlbrechtL/welle.io/archive/V1.0.tar.gz"
  sha256 "669ae5d471f723c32622cbf6ee37b66c3aefd8e02d6334b55d1fb60b3c22a883"
  head "https://github.com/AlbrechtL/welle.io.git"

  depends_on "cmake" => :build
  depends_on "qt"
  depends_on "fftw"
  depends_on "faad2"
  depends_on "librtlsdr"
  depends_on "soapysdr"
  depends_on "soapyuhd"
  depends_on "libusb"

  def install
    system "cmake", ".", "-DSOAPYSDR=TRUE", "-DBUILD_WELLE_CLI=OFF", *std_cmake_args
    system "make", "install"
  end

  test do
    system "#{bin}/welle-io", "-v"
  end
end
```

AlbrechtL, 2019-02-20:
Thanks!
Please open a pull request for the next branch with this.

rabarar, 2019-02-20:
Will do

AlbrechtL, 2019-02-21:
Thanks for the pull request https://github.com/AlbrechtL/welle.io/pull/350
Because of this I close this issue.

closed 2019-02-21 by AlbrechtL