AlbrechtL/welle.io #347: [next] Welle.io does not start

andimik, 2019-02-13:
I'm using Kubuntu 18.04. I tried both, compiled from sources and used the nightly builds, but it doesn't start

```
$ LANG=C ./apps/20190213_af8de24_Linux_welle-io-x86_64.AppImage 
2019-02-13T22:26:53.256 Debug: main: Set language "C"
2019-02-13T22:26:53.256 Debug: main: Error while loading language "C" use English "en_GB" instead
2019-02-13T22:26:53.301 Debug: Audio: ActiveState
2019-02-13T22:26:53.970 Debug: RadioController: Close device

```
No GUI showing up. The help output is working

```
$ LANG=C ./apps/20190213_af8de24_Linux_welle-io-x86_64.AppImage --help
2019-02-13T22:35:01.102 Debug: main: Set language "C"
2019-02-13T22:35:01.103 Debug: main: Error while loading language "C" use English "en_GB" instead
Usage: ./apps/20190213_af8de24_Linux_welle-io-x86_64.AppImage [options]
welle.io Help

Options:
  -h, --help               Displays this help.
  -v, --version            Displays version information.
  --dump-file <File name>  Records DAB frames (*.mp2) or DAB+ superframes with
                           RS coding (*.dab). This file can be used to analyse
                           X-PAD data with XPADxpert
  --log-file <File name>   Redirects all log output texts to a file.
  --language <Language>    Sets the GUI language according to the ISO country
                           codes (e.g. de_DE)

```

An older version `20180717_b68d488_Linux_welle-io-x86_64.AppImage` works indeed.

mpbraendli, 2019-02-14:
Maybe unrelated: Why do you set `LANG=C` ?

andimik, 2019-02-14:
Because this is an English conversation although be both speak German :wink:

mpbraendli, 2019-02-14:
Fair enough :-) If the issue occurs regardless of the setting then it is indeed irrelevant.

Do older appimages work? If you could tell us which version is the first to fail that would probably help us a lot.

andimik, 2019-02-14:
@AlbrechtL

Is there a way to download older files (next branch) than December 27th? See https://dl.bintray.com/albrechtl/welle.io/

AlbrechtL, 2019-02-17:
Old files are delete automatically because JFrog Bintray supports only 10 GB for free. If we reach the 10 GB then the account will be blocked. This happened once so I added that only last few builds will remain.

Are the following output messages the only one?
> 2019-02-13T22:26:53.256 Debug: main: Set language "C"
2019-02-13T22:26:53.256 Debug: main: Error while loading language "C" use English "en_GB" instead
2019-02-13T22:26:53.301 Debug: Audio: ActiveState
2019-02-13T22:26:53.970 Debug: RadioController: Close device

andimik, 2019-02-17:
Yes, they are.

AlbrechtL, 2019-02-17:
That's weird. Please double check it. There should be at least some shutdown messages.

andimik, 2019-02-17:
No, only `^C`...

AlbrechtL, 2019-03-03:
I don't know how to proceed.

Can you post a complete console output please.
Another options could be to delete the config file under `.config/welle.io/welle.io.conf`.

andimik, 2019-03-03:
There was really no more console output, even after hours of testing. 

Thanks, deleting (or renaming) of that file was the solution. 

I even did not know that there is a config file somewhere.

closed 2019-03-03 by andimik

AlbrechtL, 2019-03-03:
Yeah!
The saved configuration is sometimes incompatible between nightly builds.