AlbrechtL/welle.io #348: [next] Rawfile: stations not found

andimik, 2019-02-13:
I'm using latest nightly build under Windows 10 (as I cannot start it under Kubuntu, see other issue).

But it seems the programm does not correctly detect services from RAW file (I cannot test it with a DAB signal on VHF here).

[image]

[image]

[image]

[image]

[image]

Tried lots of files from all over Europe, 
both u8 and s16be file formats, 
both DAB and DAB+, 
both containing Data streams and not.

Remark: The last one is the file `sample_sync_lost.iq` from https://github.com/AlbrechtL/welle.io/issues/247

I only can tune to a station having detected its label.

AlbrechtL, 2019-02-17:
Thanks for reporting.
The issue came in between 60626a86a8eabe799755554f387b0922f05ea6e5 and ac629326c6fe5bcfb9f0464dd34eafb71f8451b6.

We are working on a fix.

mpbraendli, 2019-02-18:
Should be fixed in 9fe93d1.

AlbrechtL, 2019-02-20:
One more fix in 1fa670a04e2c8628e8770cca246df02a9e82905e.

closed 2019-02-20 by AlbrechtL