AlbrechtL/welle.io #349: SoapySDR/SDplay RSP support

Twilight-Logic, 2019-02-19:
I downloaded this today and I seem to be having problems getting it to work.

On Windows, the player does not recognize the RSP radio. It works find with an RTLSDR though.

On Linux, I get no indication that the RSP radio has not been recognized, but selecting scan for stations crashes the application. I cannot see any indication of whether the radio is connected or not so cannot provide any further information. The radio is recognized OK in SoapySDRUtil.

> $ SoapySDRUtil -info
> ######################################################
> ##     Soapy SDR -- the SDR abstraction library     ##
> ######################################################
> 
> Lib Version: v0.7.1-g7bb9ca15
> API Version: v0.7.1
> ABI Version: v0.7
> Install root: /usr/local
> Search path:  /usr/local/lib/SoapySDR/modules0.7
> Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so (0.3.0-31a3a36)
> Available factories... sdrplay
> Available converters...
>  -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CS32 -> [CS32]
>  -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CU16 -> [CF32, CS16, CS8]
>  -   CU8 -> [CF32, CS16, CS8]
>  -   F32 -> [F32, S16, S8, U16, U8]
>  -   S16 -> [F32, S16, S8, U16, U8]
>  -   S32 -> [S32]
>  -    S8 -> [F32, S16, S8, U16, U8]
>  -   U16 -> [F32, S16, S8]
>  -    U8 -> [F32, S16, S8]

mpbraendli, 2019-02-20:
The AppImage doesn't contain any SoapySDR Device drivers, which renders the SoapySDR input useless. You should be able to get it to work if you compile it yourself. Sorry about that, I think we will remove SoapySDR from the AppImage in the future.

Please reopen the ticket if you encounter an issue with your device if you compile welle.io yourself.

Twilight-Logic, 2019-02-21:
Not sure what to make of this response. The device is promoted as supporting SoapySDR devices?
Would it not be better to add the API (drivers) for the SDRplay radios?

I did attempt to compile this myself, but after a 10Gb QT download it would still not compile. I got issues with QWT which I could not resolve. I couldn't get past this point:

> $ cmake ..
> -- Build type not specified: defaulting to release.
> CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
>   Could not find a package configuration file provided by "Qt5Quick" with any
>   of the following names:
> 
>     Qt5QuickConfig.cmake
>     qt5quick-config.cmake
> 
>   Add the installation prefix of "Qt5Quick" to CMAKE_PREFIX_PATH or set
>   "Qt5Quick_DIR" to a directory containing one of the above files.  If
>   "Qt5Quick" provides a separate development package or SDK, be sure it has
>   been installed.
> Call Stack (most recent call first):
>   CMakeLists.txt:45 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/johnc/src/WelleIO/welle.io-2.0-beta1/build/CMakeFiles/CMakeOutput.log".

mpbraendli, 2019-02-21:
The point of SoapySDR is to be an abstraction between applications and the SDR devices, so that we as application developers can focus on the actual application, and the user can choose the device by installing the corresponding Soapy Device driver.

But the AppImage bundles all these things together, and so we end up having to decide what device drivers we include. That should be the user's choice! I am not aware of a way to get the SoapySDR inside the AppImage use the SoapySDR Device drivers that the user has on his system.

This is why I believe AppImage and SoapySDR don't mix well. It's quite unfortunate.

Twilight-Logic, 2019-02-21:
Yes, unfortunate, but I think I understand what you mean. This obviously leaves compiling from sources as the only option. I really don't like the idea of having to download 10Gb of QT10 stuff on my system. The current QT9.x stuff takes nowhere near this amount of space (about 200-300Mb). It is possible to deselect various modules in the QT over-the-internet install, so is there a minimum configuration I should use or do I need to install all of it?

AlbrechtL, 2019-02-21:
>  Would it not be better to add the API (drivers) for the SDRplay radios?

I asked SDRplay Ltd. for support of SDRplay inside welle.io and there were not interested. That's the reason why there is no direct support of SDRplay.

I added the SoapySDR support to the AppImage because I expected that SoapySDR will use the system libs if available. Now I see that this is not working.
There are two options
1. Remove SoapySDR from the AppImage
2. Add all SoapySDR libs to the AppImage

@mpbraendli What do you think?

@Twilight-Logic For compiling welle.io please follow the https://github.com/AlbrechtL/welle.io/blob/next/README.md and use the next branch.

mpbraendli, 2019-02-25:
I'm in favour of removing SoapySDR from the AppImage, unless we find a way to use system soapy device drivers.

Twilight-Logic, 2019-03-02:
I appreciate the dilemma and its a shame they were not interested. I will experiment some more to see if I can get anything working and will let you know if I have any success.

AlbrechtL, 2019-04-19:
Any update here?

Reopened to track it.

mpbraendli, 2019-04-23:
No updates from my side. Do you remove Soapy from the AppImage?

AlbrechtL, 2019-04-25:
> Do you remove Soapy from the AppImage?

Not yet.

chjohn22, 2019-05-06:
Maybe we must apply some pressure to the developers of SDRplay. They still do not get it, that a successful SDR does not only need superb hardware but a good support of (opensource)-Software/drivers.

SDRplay did most of the development for Windows, now they slowly begin to take care with driver support for Linux, but most of their code still is Windows only and closed source.

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL