AlbrechtL/welle.io #352: [welle.io-cli] -p program bug

mickaelmonsieur, 2019-02-24:
Hi all,

I detected a problem with the "-p" option
It seems that the program is trying to initialize the program (Could not tune to Musiq3) while the scan of the channel is not finished yet. (Ensemble name id: 6106 after)

Debug:

```
./welle-cli -c 6B -p Musiq3
Hello this is welle-cli v2.0-beta1-49-g2361f2d
InputFactory:Input device:auto
RTL_SDR:Open rtl-sdr
RTL_SDR:Found 1 devices. Uses the first working one
Found Rafael Micro R820T tuner
RTL_SDR: Opening rtl-sdr device0
[R82XX] PLL not locked!
RTL_SDR: Supported gain values29
RTL_SDR: gain 49.6
RTL_SDR: gain 48
RTL_SDR: gain 44.5
RTL_SDR: gain 43.9
RTL_SDR: gain 43.4
RTL_SDR: gain 42.1
RTL_SDR: gain 40.2
RTL_SDR: gain 38.6
RTL_SDR: gain 37.2
RTL_SDR: gain 36.4
RTL_SDR: gain 33.8
RTL_SDR: gain 32.8
RTL_SDR: gain 29.7
RTL_SDR: gain 28
RTL_SDR: gain 25.4
RTL_SDR: gain 22.9
RTL_SDR: gain 20.7
RTL_SDR: gain 19.7
RTL_SDR: gain 16.6
RTL_SDR: gain 15.7
RTL_SDR: gain 14.4
RTL_SDR: gain 12.5
RTL_SDR: gain 8.7
RTL_SDR: gain 7.7
RTL_SDR: gain 3.7
RTL_SDR: gain 2.7
RTL_SDR: gain 1.4
RTL_SDR: gain 0.9
RTL_SDR: gain 0
OFDM-processor:start
Start RTLSDR thread
Wait for sync
ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
Service list
Could not tune to Musiq3
**** Please enter programme name. Enter '.' to quit.
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: Found sync (coarseCorrector: 0; fineCorrector: 0 after 63 frames)
Ensemble name id: 6106
New Service: 0x6359
New Service: 0x6366
New Service: 0xe0605006
New Service: 0x6353
New Service: 0x6360
New Service: 0x6355
New Service: 0x6217
New Service: 0x6482
New Service: 0x63bd
{"TII":{"comb":1,"delay":434,"delay_km":63.57421875,"error":255.0,"pattern":6}}
{"TII":{"comb":1,"delay":111,"delay_km":16.259765625,"error":273.0,"pattern":6}}
{"TII":{"comb":1,"delay":476,"delay_km":69.7265625,"error":259.0,"pattern":6}}
^C
```
It seems to me that a modification should be made in the following file: src/welle-cli/welle-cli.cpp around line 514. 

We check with a 'while' if 'service_to_tune' is not empty, but we should rather check that rx.getServiceList () is not empty. No ?

I do not have the knowledge of C++ to patch but I'm pretty sure it's simple to fix :)

Best regards,
Mickael

mickaelmonsieur, 2019-02-24:
Important Information: I have this problem only with the next branch and not the master.

mpbraendli, 2019-02-25:
Thanks for reporting this issue. I'm afraid I won't have time to look into that anytime soon.

You have done some source code analysis already, don't be afraid to modify things and see if you can improve it yourself!

tenzap, 2019-08-14:
Hello,
I have the same kind of issue

mpbraendli, 2019-08-19:
Should be fine now.

commit 97a214ca (2019-08-19): Add additional wait time in welle-cli -p

Fixes #352
closed 2019-08-19 by mpbraendli