AlbrechtL/welle.io #353: [welle-gui] No text in GUI

deltafox1002, 2019-02-24:
When I start the App it show the frame but no text. System is Linux Mint 19.1 Cinnamon.

AlbrechtL, 2019-03-03:
Please post a screen shot and the console output.

AlbrechtL, 2019-05-03:
Any news here?

cryptik, 2019-08-23:
I have a similar issue.  The application finds the RTL_SDR and it appears to be working correctly, but the GUI is blank (for the most part).  Here is a screenshot of the GUI as well as the console window.  I get the same results running the appimage as well as building it with QtCreator.

When building the app, I could not find Qt5.10 so I built it with Qt5.11... Qt5.12 also produces the same results.

[image]
[image]

cryptik, 2019-08-23:
I see the warnings regarding visibility and the createObject issue.  I don't know Qt very well...but I will try and look into it more.  It must be some issue between Qt and Ubuntu 18.04 or maybe Qt 5.11.  It does not appear to be a common issue.

cryptik, 2019-08-23:
Just for completeness, this is what the display looks like now...after increasing the amount of video memory assigned:

[image]

The console is reporting an "Info: ofdm-processor: SyncOnNull failed" when running a station scan, but I imagine this is what it says, and info message.

[image]

AlbrechtL, 2019-08-24:
Thanks for the update and the analysis!

closed 2019-08-24 by AlbrechtL