AlbrechtL/welle.io #356: welle-cli web stream url - allow to select channel

LeoSchleicher, 2019-03-25:
I want to use welle-cli instance vie other program (USB Player pro). This app allows playing via http streams, like many other apps do. But in case of welle-cli web interface I can provide in stream URL only ID, assuming my rtl-sdr device already tuned to apropriate DAB-channel, ie /mp3/0x1111 

Is it already implemented or teoretically possible to control channel via stream URL? ie /mp3/11c/0x1111. This allows using welle-cli as local broadcast server with complete remote control.

mpbraendli, 2019-03-28:
Hi Leo,

Channel switching takes some time, and you first need to select a channel and wait for synchonisation and ensemble composition decode before you can ask for specific service. To implement both channel switching and programme selection through one URL is doable, but would break some other uses.

For example, if user A listens to one service, and user B requests a service on another ensemble, he would unknowingly disrupt user A.

There is an API endpoint for switching the channel, as you are probably aware of if you looked into the web UI.

I hope this answers your question!

closed 2019-03-28 by mpbraendli

LeoSchleicher, 2019-03-29:
Hi,
thank you for answer. Using webUI is also possible, but there is some problems. As I use another App to stream radio, I must in WebUI or in my custom App first tune the channel, then in player-App connect to some program with url to this program. Also, I must first define in player all programs of my interests (urls). But I cannot just switch between them, as I must first tune the channel in web or another app.
Alternatively it may be acceptable to have just one stream url, that user can tune externally. This will be also good, as in car (my case) available channels and programs can vary dependent on location, so just recreate the list of available channels for player-app is tricky. But if I can provide just one streaming url in player and then tune channel/program via API - this can be better solution.

Unfortunately, I must rely only to one predefined player app, as it and only it can work with my audio hardware.

LeoSchleicher, 2019-03-29:
Just looked at code. I am not so good in cpp, but the Idea is:

1. Add /program HTTP - action, that sets and stores in global variables some program ID
2. by send_mp3 method also allow calls without ID, and if the global program set, just stream mp3 with given program (as fallback - ie just use first program from channel)

this solution seems easy, but can help me a lot. I don't know if this can be useful for anyone else.

mpbraendli, 2019-04-01:
Ok, I understand your constaints a bit better. Indeed having both `/mp3/<sid>` and `/mp3/<channel>/<sid>` would not break the current usage.

I don't have time to implement and test this, but you can try implementing it yourself. Add a regex to match the new url, and use the `retune` function to switch channel. You will have to wait a bit before you can request the mp3 data though, the receiver needs to sync first.

Let me know how you get along.

LeoSchleicher, 2019-04-01:
Thank you! I've made already some changes:

1. allowed /mp3 call without program id
2. by such calls I stream id stored in global property
3. if stored id isn't defined yet, I stream first available program from the current channel
4. implemented switching between programs within channel without breaking currently streaming mp3-stream - just detaching and reattaching program handlers to the socket.

This works, but not so stable, as I am not an experienced cpp coder. Now I try to implement channel switching without breaking currently streamed mp3.

comcloudway, 2022-06-04:
Any updated on this?
I failed to find your welle.io fork @LeoSchleicher assuming you ever uploaded it.
And thus I decided to write the implementation myself.
The channel switching seems to be working and I'm using a while loop to wait for the sync to succeed.

Unfortunately I can't get it to work reliably, out of ten times it worked like two times the other 8 tries it `Could not setup mp3 sender for 54230: map::at`.
As far as I can tell the map is missing an entry for the channel Id.

As you can see by checking my code ([regex](https://github.com/comcloudway/welle.io/blob/a8b8d8c19bce081d1c9e6acb8ad6815157f68070/src/welle-cli/webradiointerface.cpp#L516) and [send_mp3](https://github.com/comcloudway/welle.io/blob/a8b8d8c19bce081d1c9e6acb8ad6815157f68070/src/welle-cli/webradiointerface.cpp#L856) ), I placed the while loop in the `send_mp3` function and oddly enough requesting `/mp3-channel/<channel>/<program>` ends up changing the channel but still fails. 
Calling `/mp3/<program>` immediately after the other request fails seems to play fine though.

Also I have noticed that `retune` seems to re-sync even if the channel stays the same which causes unnecessary delay

comcloudway, 2022-06-04:
### Update
I fixed the map issue by manually creating a `WebProgrammeHandler`.
Everything seems to work fine in the latest commit

To test it you can clone my fork (link above)
and visit `/mp3-channel/<channel>/<program>`
it will take some time to load as it has to wait for the scan to finish, but it should afterwards.

I'll work through the contribution guidelines and maybe I'll get a PR ready today - will keep you updated