AlbrechtL/welle.io #359: [rtl-tcp] 2.0 Beta 1: No GUI, if TCP connection cannot be established

dl9sec, 2019-04-01:
When welle.io is configured for rtl-tcp and the connection cannot be established, the GUI doesn't come up. So it is not possible to change the source e.g. to direct USB connection of a rtl-sdr device or to get a diagnosis (e.g. "No connrction to rtl-tcp.").

AlbrechtL, 2019-04-19:
Which version on which OS do you are using?

dl9sec, 2019-04-22:
2.0 Beta 1 on a Windows 10.

AlbrechtL, 2019-04-22:
Can you try to use the command line option `--log-file yourfile.txt`? Please post the resulting file into here.

dl9sec, 2019-04-23:
Here it is:
2019-04-23T07:04:00.458 Debug: main: Version: "2.0-Beta 1 Git: f00230c"
2019-04-23T07:04:01.122 Debug: RadioController: Close device
2019-04-23T07:04:01.122 Debug: RadioController: AGC off
2019-04-23T07:04:01.122 Debug: RadioController: Play channel: "9D" station: "SWR3            "
2019-04-23T07:04:01.137 Info: RTL_TCP_CLIENT: Try to connect to server 192.168.100.40:2832
2019-04-23T07:04:01.349 Debug: Audio: ActiveState
2019-04-23T07:04:01.363 Debug: Audio: Volume 1
2019-04-23T07:04:21.458 Info: Could not connect
2019-04-23T07:04:21.458 Info: RTL_TCP_CLIENT: Could not connect to server
2019-04-23T07:04:21.458 Debug: RadioController: Radio device is not ready or does not exist.
2019-04-23T07:04:21.477 Debug: RadioController: Tune to channel "9D" -> 208.064 MHz
2019-04-23T07:04:21.578 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS threshold: -1
2019-04-23T07:04:21.578 Info: OFDM-processor:start
2019-04-23T07:04:21.583 Info: RTL_TCP_CLIENT: Try to connect to server 192.168.100.40:2832

AlbrechtL, 2019-04-25:
Can you try the latest nightly built please?
https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

dl9sec, 2019-04-29:
Can confirm the fix. Working well with the latest nightly built :-)
Great stuff! Thank you...

closed 2019-04-29 by dl9sec