AlbrechtL/welle.io #360: Multiple receivers

EPO35F, 2019-04-10:
Hello, I have installed to receivers, start 2 welle-io.exe, and only the first one works.
Any hints how to have several ones working ?
I am mainly interested by the data within DAB
I have Nooelec receiver and RTLSDR RTL2832U V3 ones

AlbrechtL, 2019-04-19:
Welle.io uses automatically the first device. Which is the use cases use two devices at the same time?

ariendj, 2019-05-13:
> Welle.io uses automatically the first device. Which is the use cases use two devices at the same time?

I'd like to run one Welle.io web interface for every mux that I can receive. This way I'd be able to stream any number of DAB channels I receive to any number of network devices.

Also I want to try and pass complete DAB muxes to TVHeadend. This would allow listening, recording and timeshifting any number of DAB channels on my multiple Kodi frontends.

AlbrechtL, 2019-05-23:
If you like you can start implementing these features. We are always open for pull request.

AlbrechtL, 2019-08-10:
Closed because it is duplicated. See issue https://github.com/AlbrechtL/welle.io/issues/95 please.

closed 2019-08-10 by AlbrechtL