AlbrechtL/welle.io #362: [Ubuntu 16.04] 20190419_4f18916_Linux_welle-io-x86_64.AppImage not loading properly

erikdsgh, 2019-04-20:
For your information:
When starting version 20190419_4f18916_Linux_welle-io-x86_64.AppImage only 8.7 Mb is loaded in memory. The icon in the top bar appears but nothing else.
With version 20190411_21ef78b_Linux_welle-io-x86_64.AppImage that works ok about 65 Mb is loaded into memory.

AlbrechtL, 2019-04-21:
Thanks for reporting!

Please try to delete the config files and try start welle.io again.

Please see the FAQ to see where the config files are stored. https://www.welle.io/faq#where-are-the-settings-stored

erikdsgh, 2019-04-21:
I have done that.
No change. Still only 8.7 Mb loaded and only icon in top bar but no wello.io screen opened.
No new config file is made.

AlbrechtL, 2019-04-21:
Ah I see.
The issue was that I used Qt 5.12 / Qt 5.13 for the development. But we deliver welle.io with Qt 5.11. It seems that Qt 5.11 doens't support [ECMAScript_2015](https://en.wikipedia.org/wiki/ECMAScript#6th_Edition_-_ECMAScript_2015).

Fixed in https://github.com/AlbrechtL/welle.io/commit/0819e83f0d56367a7ce07459990ee79d955d243b.

erikdsgh, 2019-04-21:
OK 0819e83 works.
Thanks for the very rapid reaction and best regards.

closed 2019-04-22 by AlbrechtL