AlbrechtL/welle.io #363: welle-cli on Raspberry Pi - sound slowed down on mono program

NottsIan, 2019-04-24:
Hello, I've successfully built welle-cli (2.0 beta 1) on Raspberry Pi 3B+ running Raspbian and it works really well on stereo programs. With a mono program, the sound plays at half speed.

As a short-term workaround I've edited DecoderAdapter::PutAudio to disable the mono/stereo test and although this works well I don't think it's the correct solution. It looks like something else is also doubling the data to make two identical channels, could this be something in the Alsa configuration on the Pi or is there something else I've missed?

Thanks
Ian.

AlbrechtL, 2019-04-25:
Thanks for reporting.
Can you try the latest code from the next branch https://github.com/AlbrechtL/welle.io/tree/next please?

NottsIan, 2019-04-25:
I'm going away for a couple of days but as soon as I get back I'll try it.

NottsIan, 2019-05-01:
The latest code works fine. I can see in welle-cli.cpp where the problem has been fixed.

AlbrechtL, 2019-05-03:
Mh, interesting. We had a similar issue in Nov, 2018 (https://github.com/AlbrechtL/welle.io/issues/316). But the fix is included in 2.0 beta1. From 2.0 beta1 to now we didn't change the audio code as far as I know.

For now I'm closing it.

closed 2019-05-03 by AlbrechtL