AlbrechtL/welle.io #367: Remove hide() when running on Gnome Desktop

meeuw, 2019-04-27:
Hi, I actually don't know how to fix this properly but Gnome Desktop doesn't have a "System Tray" and when welle.io's main window is minimized it's impossible to open the window again (because it's hidden from the "taskbar").

This can be disabled by removing `onMinimizeWindow: hide()` from `src/welle-gui/QML/MainView.qml` but that would kill the user experience for users with a System Tray.

The best suggestion I can think of is to only hide the main window when the welle.io icon is clicked in the system tray.

AlbrechtL, 2019-04-28:
Good point.

The best would be if Qt can detect during run time if there is a sys tray. Can you check if Qt provides a API for this?

meeuw, 2019-04-28:
Hmm not so sure if there is an Qt API. I would settle for checking the environment variable `DESKTOP_SESSION` and if it's set to `"gnome"` don't hide the window.

AlbrechtL, 2019-04-29:
Good idea with the environment variables.

The things are a little bit more complicated. I'm using Ubuntu with it's Gnome variant and a system tray.

meeuw, 2019-04-29:
Hmm, seems there's a Qt API after all...

I've created the following test:
```c++
#include <QtWidgets/QSystemTrayIcon>
#include <QtWidgets/QApplication>

int main(int argc, char **argv) {
    QApplication a(argc, argv);
    printf("QSystemTrayIcon::isSystemTrayAvailable: %i\n", QSystemTrayIcon::isSystemTrayAvailable());
}
```

```bash
g++ main.cpp -I/usr/include/qt5/ -fPIC -lQt5Core -lQt5Widgets ; ./a.out
```

outputs:
```
$ g++ main.cpp -I/usr/include/qt5/ -fPIC -lQt5Core -lQt5Widgets ; ./a.out 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QSystemTrayIcon::isSystemTrayAvailable: 0
```

Could you test this on your system?

AlbrechtL, 2019-04-30:
Thanks for figuring this out.

I added this into https://github.com/AlbrechtL/welle.io/commit/426331163e4509302203495aebfe07eb1fb07acd.
https://github.com/AlbrechtL/welle.io/blob/426331163e4509302203495aebfe07eb1fb07acd/src/welle-gui/gui_helper.cpp#L246-L253

Please test the latest nightly build or build it from the next branch.

meeuw, 2019-04-30:
:+1: works for me!

closed 2019-05-01 by AlbrechtL