AlbrechtL/welle.io #368: Next/previous station

joural, 2019-05-06:
I'd like to change between stations with rotary encoder, is there some way I could implement this in welle?

AlbrechtL, 2019-05-07:
Can you describe your project more in detail please?

joural, 2019-05-07:
Yeah replace car radio with rpi running open auto where welle.io is included, have one rotary encoder for volume control and other to simulate tunning

joural, 2019-05-08:
If there would be keys to play next/ previous station I think I could simulate keypresses with python script

jay-to-the-dee, 2019-05-14:
Tbf, up/down or left/right arrow key channel navigation would be pretty intuitive and useful in the Qt GUI. Maybe PgUp/PageDown could change mux? That's a similar principle to how my car DAB radio does it.

AlbrechtL, 2019-05-23:
I think @joural and @jay-to-the-dee you are mentioned two different things.
1. The question is how to implemented an interface to control welle.io remotely
2. Hot keys

I think the hot keys could be interesting feature for a future version. The remote interface is a little bit to specialized.

joural, 2019-05-23:
Hot keys is fine, I'm looking forward to future version. Shall I close the issue, or leave it open? Thank you

jay-to-the-dee, 2019-05-24:
@AlbrechtL hot keys sounds like it would be useful for us both then :)

AlbrechtL, 2019-05-30:
To implement hot key makes sense. For that reason I leave this ticket open. Please keep in mind that the priority is low for the feature request. But we are open for pull requests!

AlbrechtL, 2019-12-06:
To remote control welle.io I'm thinking of implementing a JSON API (https://doc.qt.io/qt-5/json.html) via TCP.
What do you think?

joural, 2019-12-06:
That would work for me 👍

tenzap, 2021-02-27:
For now you have some hotkey support in next (see #628 for details). Shortcuts are also in the README of next

AlbrechtL, 2021-03-27:
I'm closing this issue because https://github.com/AlbrechtL/welle.io/pull/628 is merged.

closed 2021-03-27 by AlbrechtL