AlbrechtL/welle.io #369: AirSpy not usable on Linux Ubuntu 18.04

amontefusco, 2019-05-17:
Trying to use AirSpy receiver on Ubuntu 18.04 with welle.io, I receive the following error:

```
~/Downloads/welle.io_2.0-Beta2_Linux_x86_64.AppImage 
2019-05-17T13:58:12.937 Debug: Audio: ActiveState
2019-05-17T13:58:13.091 Debug: main: Set language "en_US"
2019-05-17T13:58:13.091 Debug: main: Error while loading language "en_US" use English "en_GB" instead
2019-05-17T13:58:13.114 Debug: Apply settings initially
2019-05-17T13:58:13.114 Debug: RadioController: Close device
2019-05-17T13:58:13.118 Info: Airspy:Open airspy
2019-05-17T13:58:13.127 Info: Airspy:airspy_set_samplerate() failed:AIRSPY_ERROR_LIBUSB(-1000)
2019-05-17T13:58:13.128 Info: InputFactory:Error while opening device "2".
```

The same device is working fine both with its utilities

```
airspy_info 
airspy_lib_version: 1.0.9

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc3-0-g4d9e874 2015-01-09
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x440464C838269F4F
Supported sample rates:
	10.000000 MSPS
	2.500000 MSPS
Close board 1
```

```
airspy_rx -f 100 -a 1 -w /dev/null
Receive wav file: AirSpy_20190517_140226Z_100000kHz_IQ.wav
Device Serial Number: 0x440464C838269F4F
Stop with Ctrl-C
Streaming at 2.500 MSPS
Streaming at 2.500 MSPS
Streaming at 2.500 MSPS
Streaming at 2.500 MSPS
Streaming at 2.500 MSPS
^CCaught signal 2

User cancel, exiting...
Total time: 5.6056 s
Average speed 2.5001 MSPS IQ
done
andrew@andrewp:~/tmp/sdrangel/build$ airspy_rx -f 100 -a 0 -w /dev/null
Receive wav file: AirSpy_20190517_140235Z_100000kHz_IQ.wav
Device Serial Number: 0x440464C838269F4F
Stop with Ctrl-C
Streaming at 10.001 MSPS
Streaming at 10.002 MSPS
Streaming at 9.999 MSPS
Streaming at 10.000 MSPS
Streaming at 9.995 MSPS
Streaming at 10.000 MSPS
Streaming at 10.000 MSPS
Streaming at 10.001 MSPS
^CCaught signal 2

User cancel, exiting...
Total time: 8.0414 s
Average speed 10.0005 MSPS IQ
done

```

and other SDR programs (e.g., SDR Angel).

amontefusco, 2019-05-17:
Forgot to say that, on the same system, AirSpy Mini is working fine, but, occasionally, welle.io receives the same error:

```
Airspy:airspy_set_samplerate() failed:AIRSPY_ERROR_LIBUSB(-1000)
```

AlbrechtL, 2019-05-23:
It looks like an airspy library specific issue. What did Google say to `AIRSPY_ERROR_LIBUSB(-1000)`?

andimik, 2019-05-23:
Do you have an Airspy or an Airspy mini? The later does not allow 10 MS as the bandwidth cannot handle it.

amontefusco, 2019-05-24:
> Do you have an Airspy or an Airspy mini? 

I have both.
The one that doesn't work is the Airspy regular size.
The Mini is working properly both with 1.0 and 2.0

> But I wonder why you wanted to use such a big sampling rate. As DAB is a 1,7 MHz signal, 2048 kS are enough.

Please note that the Airspy is not detected at all with 2.0 (it is working fine on the same PC, USB cable, USB port )  with the 1.0.

andimik, 2019-05-27:
I don't have a problem under Kubuntu 18.04 and the latest welle-io 2.0 Beta 2 (compiled from the sources), Qt-version 5.11.1

I also tried https://bintray.com/albrechtl/welle.io/download_file?file_path=20190505_9f1180f_Linux_welle-io-x86_64.AppImage , which also works with my Airspy Mini.

Please try to update your very old firmware. I'm using rc10, you are still having rc3.

https://github.com/airspy/airspyone_firmware/releases

```
$ airspy_info 
airspy_lib_version: 1.0.9

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-0-g946184a 2016-09-19
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0x04A464C8377B3D0B
Supported sample rates:
        6.000000 MSPS
        3.000000 MSPS
Close board 1

```

```
$ ./welle-io 
QML debugging is enabled. Only use this in a safe environment.
2019-05-27T21:29:47.875 Debug: Audio: ActiveState
2019-05-27T21:29:48.231 Debug: main: Set language "de_DE"
2019-05-27T21:29:48.255 Debug: Apply settings initially
2019-05-27T21:29:48.255 Debug: RadioController: Close device
2019-05-27T21:29:48.260 Info: Airspy:Open airspy
2019-05-27T21:29:48.278 Debug: RadioController: AGC on
2019-05-27T21:29:48.278 Debug: Audio: Volume 1
2019-05-27T21:29:48.279 Info: Airspy: Set bias tee to 0
2019-05-27T21:29:48.427 Warning: file:///home/andreas/Qt/5.11.1/gcc_64/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window'
2019-05-27T21:29:48.482 Debug: os: linux
2019-05-27T21:29:48.482 Debug: desktopAvailableWidth: 1366
2019-05-27T21:29:48.482 Debug: desktopAvailableHeight: 732
2019-05-27T21:29:48.483 Debug: orientation: 2
2019-05-27T21:29:48.483 Debug: devicePixelRatio: 1
2019-05-27T21:29:48.483 Debug: pixelDensity: 3.9751024219785513
2019-05-27T21:30:01.410 Debug: RadioController: Tune to channel "12C"
2019-05-27T21:30:01.418 Debug: RadioController: Tune to channel "12C" -> 227.36 MHz
2019-05-27T21:30:01.859 Info: New Receiver Options: TII: 1 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2019-05-27T21:30:01.859 Info: OFDM-processor:start
2019-05-27T21:30:01.933 Info: ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
2019-05-27T21:30:01.980 Info: ofdm-processor: SyncOnPhase failed
2019-05-27T21:30:02.476 Debug: RadioController: ID of ensemble: 20489
2019-05-27T21:30:04.309 Info: ofdm-processor: SyncOnPhase failed
2019-05-27T21:30:04.587 Info: ofdm-processor: Found sync (coarseCorrector: 0; fineCorrector: 0 after 27 frames)
2019-05-27T21:30:04.682 Info: ofdm-processor: Lost coarse sync (coarseCorrector: 30000; fineCorrector: 132)
2019-05-27T21:30:05.067 Info: ofdm-processor: SyncOnPhase failed
2019-05-27T21:30:05.356 Info: ofdm-processor: Found sync (coarseCorrector: 30000; fineCorrector: 132 after 6 frames)
2019-05-27T21:30:05.461 Debug: RadioController: Found service  5283 "Deejay 30 Songs "
2019-05-27T21:30:06.152 Debug: RadioController: Found service  5288 "Kc2Test         "
2019-05-27T21:30:06.185 Debug: RadioController: Found service  52CC "RMALB.IA        "
2019-05-27T21:30:06.342 Debug: RadioController: Found service  5264 "* RDS *         "
2019-05-27T21:30:06.385 Debug: RadioController: Found service  5233 "M DUE O         "
2019-05-27T21:30:06.421 Debug: RadioController: Found service  5280 "Kc1Test         "
2019-05-27T21:30:06.468 Debug: RadioController: Found service  5219 "Capital         "
2019-05-27T21:30:06.504 Debug: RadioController: Found service  5281 "* RDS * Relax   "
2019-05-27T21:30:06.552 Debug: RadioController: Found service  5289 "Kc3Test         "
2019-05-27T21:30:06.585 Debug: RadioController: Found service  5245 "Radio 24        "
2019-05-27T21:30:06.623 Debug: RadioController: Found service  5214 "Deejay          "
2019-05-27T21:30:06.669 Debug: RadioController: Found service  5215 "R101            "
2019-05-27T21:30:06.702 Debug: RadioController: Found service  51CC "Radio Maria     "
2019-05-27T21:30:06.751 Debug: RadioController: Found service  5282 "Radio 24 +1     "
2019-05-27T21:30:06.785 Debug: RadioController: Found service  5287 "M DUE O DANCE   "
2019-05-27T21:30:06.823 Debug: RadioController: Found service  5286 "105 DAB         "
2019-05-27T21:30:06.869 Debug: RadioController: Found service  5285 "Capital Funky   "
2019-05-27T21:30:06.903 Debug: RadioController: Found service  5210 "Radio Radicale  "
2019-05-27T21:30:09.257 Info: ofdm-processor: Lost coarse sync (coarseCorrector: -4000; fineCorrector: 120)
2019-05-27T21:30:09.605 Info: ofdm-processor: SyncOnPhase failed
2019-05-27T21:30:09.967 Info: ofdm-processor: SyncOnPhase failed
2019-05-27T21:30:10.250 Info: ofdm-processor: Found sync (coarseCorrector: -4000; fineCorrector: 120 after 11 frames)
```

[image]

amontefusco, 2019-05-27:
You are right: the welle 2.0 are sensitive to Airspy ONE (not Mini !) firmware release.
Now it works with both 1.'0 and 2.0.
Many thanks for the support.

closed 2019-05-27 by amontefusco