AlbrechtL/welle.io #372: [macOS, Homebrew] -- Configuring incomplete

Ivesuwe, 2019-06-14:
on MacOS 10.14.5

$ git -C "$(brew --repo homebrew/core)" fetch --unshallow
$ brew tap pothosware/homebrew-pothos
$ brew tap dholm/homebrew-sdr #other sdr apps
$ brew update
$ brew tap AlbrechtL/welle_io https://github.com/AlbrechtL/welle.io
$ brew install AlbrechtL/welle_io/welle.io

...
-- Configuring incomplete, errors occurred!
See also "/tmp/welle.io-20190614-30636-193fikv/welle.io-2.0-beta1/CMakeFiles/CMakeOutput.log".

If reporting this issue please do so to (not Homebrew/brew or Homebrew/core):
  albrechtl/welle_io

/usr/local/Homebrew/Library/Homebrew/utils/github.rb:258:in `raise_api_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::ValidationFailedError)
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:212:in `open_api'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:357:in `search'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:265:in `search_issues'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:278:in `issues_for_formula'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:374:in `fetch_issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:370:in `issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:424:in `dump'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:141:in `rescue in <main>'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:38:in `<main>'
...

CMakeOutput.log does not exist

AlbrechtL, 2019-07-01:
I don't have a Mac so I can't help you. Hopefully somebody with a Mac can help.

MarcGodard, 2019-07-12:
I have a similar error:

```
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
CMake Error at /usr/local/Cellar/cmake/3.14.5/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find MPG123 (missing: MPG123_LIBRARIES MPG123_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.14.5/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindMPG123.cmake:28 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:69 (find_package)

-- Configuring incomplete, errors occurred!
See also "/tmp/welle.io-20190712-44966-5vw4ll/welle.io-2.0-beta1/CMakeFiles/CMakeOutput.log".

If reporting this issue please do so to (not Homebrew/brew or Homebrew/core):
  albrechtl/welle_io

/usr/local/Homebrew/Library/Homebrew/utils/github.rb:258:in `raise_api_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::ValidationFailedError)
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:212:in `open_api'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:357:in `search'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:265:in `search_issues'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:278:in `issues_for_formula'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:374:in `fetch_issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:370:in `issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:424:in `dump'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:141:in `rescue in <main>'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:38:in `<main>'
```

AlbrechtL, 2019-07-17:
Can you try the next branch https://github.com/AlbrechtL/welle.io/tree/next?

AlbrechtL, 2019-08-10:
Closed because issue https://github.com/AlbrechtL/welle.io/issues/333 handles it.

closed 2019-08-10 by AlbrechtL