AlbrechtL/welle.io #373: [Android] App ARMv7 doesn't start

Mikescotland, 2019-06-21:
On nexus 7 with marshmallow version ARMv7 doesn't start. Stays on initial icon forever.

AlbrechtL, 2019-07-01:
Which version did you use?

Mikescotland, 2019-07-01:
welle.io_2.0-Beta2_Android_ARMv7.apk

Of course version 1 from Google play works but would like to have version
2! Thanks.

AlbrechtL, 2019-07-01:
Thanks for reporting!

AlbrechtL, 2019-07-01:
Can you try to disconnect the rtl-sdr from your phone and start welle.io again?

Mikescotland, 2019-07-01:
I did it, same behaviour. Freezes on icon start screen and doesn't proceed
any further. I have to kill the process (soft button go back doesn't work
either on this screen)

AlbrechtL, 2019-07-17:
Thanks for helping! We definitely need to work on the Android version.

AlbrechtL, 2019-08-11:
We fixed some bugs related to Android.

Please try the version [20190811_Android_armv7_welle-io.apk](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/8f20ae8c246248585de2c1709b24195b1857005c/20190811_Android_armv7_welle-io.apk)?

meepiquitous, 2019-08-12:
Although I'm not OP, this new build works perfectly on my Kindle Fire HD10.

Thank you so much!

Mikescotland, 2019-08-12:
Hello
Starts OK and works. Noticed a few things though 
1. Scanning. Stuck on first frequency 5A forever, need to manually change the channel in manual mode.
2. AGC seems to be too "gentle". While in wavesink a particular weaker Station plays great, welle.io loses signal immediately and chops the sound.
3. Last station playing after exit option doesn't work (doesn't tune to the station), I need to change the station to a different channel and then come back. 
4. Welle.io doesn't sent notification as it used to do (Station name etc) and also doesn't react now on media buttons (next, previous etc). 

Thanks for good work

AlbrechtL, 2019-08-12:
Thanks for testing!

@Mikescotland Please be so kind to open for each point a new issue. This make us easy to track and to discuss the points without mixing it.

closed 2019-08-12 by AlbrechtL

meepiquitous, 2019-08-20:
@AlbrechtL could you please add your 20190811_Android_armv7_welle-io.apk on the releases page?

AlbrechtL, 2019-08-24:
Please by patient.
We are planning to release a new beta soon. Since this a free time project the thinks are not going as fast as they should be.