AlbrechtL/welle.io #377: welle-cli with rtl_tcp

haditim, 2019-07-10:
Hi,
I am loving the api in welle-cli interface. However it does not seem to be working with rtl_tcp (or I could not find it). Would there be a chance to add this to the task-list?

AlbrechtL, 2019-07-17:
It should be not so difficult to add rtl_tcp to welle-cli if you unterstand a little bit C++.

The input device it opened here
https://github.com/AlbrechtL/welle.io/blob/4d0b770fab4ad0b1f594794a9303060d3745249e/src/welle-cli/welle-cli.cpp#L394

You can change the `"auto"` to `CDeviceID::RTL_TCP`. Please take a look into the input factory class
https://github.com/AlbrechtL/welle.io/blob/4d0b770fab4ad0b1f594794a9303060d3745249e/src/input/input_factory.cpp#L79-L116

Afterwards you need to set the IP address. An example can be found here
https://github.com/AlbrechtL/welle.io/blob/4d0b770fab4ad0b1f594794a9303060d3745249e/src/welle-gui/radio_controller.cpp#L130-L131

I hope that helps.

haditim, 2019-07-17:
Thanks a lot for the reply. I'll give it a try.
What about an ETI input (from file/stdin/zmq)? adding these sources would complement this amazing tool.

AlbrechtL, 2019-07-21:
For ETI input see https://github.com/AlbrechtL/welle.io/issues/217 please.

seife, 2020-01-03:
Fixed by pr #493 ;-)

closed 2020-01-04 by haditim