AlbrechtL/welle.io #378: NooElec USB Receiver not working on Ubuntu 18.04

perlian, 2019-07-14:
Hello

I have a NooElec USB Receiver but its not working on Ubuntu 18.04
i use the appimage and it shows no receiver found

dmesg shows:
dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected

whats the problem ?

FartherOut, 2019-07-16:
Off the top of my head I believe you may have the “DVB” drivers installed for the unit.
Have a look at Nooelecs installation manual https://www.nooelec.com/store/downloads/dl/file/id/72/product/0/nesdr_installation_manual_for_ubuntu.pdf

AlbrechtL, 2019-07-17:
Did you try to install the install `librtlsdr0` lib?
```
# sudo apt install librtlsdr0
```

perlian, 2019-07-17:
Yes sure

Im using linux mint
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/AlbrechtL/welle.io/issues/378?email_source=notifications\u0026email_token=ACG7JE6M3TJLIAPFR336R4DP75NS7A5CNFSM4IDSIVDKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2GDQMA#issuecomment-512505904",
"url": "https://github.com/AlbrechtL/welle.io/issues/378?email_source=notifications\u0026email_token=ACG7JE6M3TJLIAPFR336R4DP75NS7A5CNFSM4IDSIVDKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2GDQMA#issuecomment-512505904",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

AlbrechtL, 2019-07-21:
Did you try to disable `dvb_usb_rtl28xxu` kernel module https://www.rtl-sdr.com/forum/viewtopic.php?t=111#p9292?

AlbrechtL, 2019-08-10:
Any updates here?

AlbrechtL, 2019-12-02:
Closed because of no activity.

closed 2019-12-02 by AlbrechtL