AlbrechtL/welle.io #379: Compiling Problems  lame/lame.h and other libraries not found error

Saadi2015, 2019-07-18:
In file included from ../../../welle.io/src/welle-cli/webprogrammehandler.cpp:25:0:

../../../welle.io/src/welle-cli/webprogrammehandler.h:29:23: fatal error: lame/lame.h: No such file or directory

compilation terminated.

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lfftw3f

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lfaad

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lmp3lame

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lmpg123

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lairspy

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lrtlsdr

/home/saadi/raspi/gcc-linaro-5.5.0-2017.10-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/5.5.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lSoapySDR

collect2: error: ld returned 1 exit status
[image]
[image]

AlbrechtL, 2019-07-21:
Did you follow the instructions on https://github.com/AlbrechtL/welle.io#ubuntu-linux-1804-lts?

Saadi2015, 2019-07-23:
Hello Albrecht,

yes I did follow the description you linked on ubuntu-16.04.4 and Raspbian Buster..
Can I compile it on Raspbian directly without cross-compilation?
How do I do that ?

Thanks for the fast reply!

AlbrechtL, 2019-08-10:
Any updates here?

Saadi2015, 2019-08-19:
I still did not have any time to try it out, but I will post updates, when I'll come to it.

closed 2019-10-21 by mpbraendli