AlbrechtL/welle.io #381: Missing station info / radiotext etc

marrem, 2019-07-28:
I'm on MacOS 10.14.6.
I managed to build the project. It scans and tunes perfectly.
But the window is rather empty. I doesn't show all the station information I see here:

https://www.welle.io/

Mine looks like:
[
[image]
](url)

AlbrechtL, 2019-08-10:
Can you post the command line output, please?

AlbrechtL, 2019-08-11:
Can you try the latest version from the next branch please?

I fixed a bug in https://github.com/AlbrechtL/welle.io/commit/3e74e70c0c73713de29f6de39dcc9bd0b1f041d6 and I would like to know if this was your issue.

marrem, 2019-08-12:
I have just built the 'next' branch which contains the mentioned commit. I has built correctly. I am currently on holiday in France. I'm in a small village 'in the middle of nowhere'. No DAB stations here... The screen looks more promising. I can enable 'expert settings' and add a few windows, like 'spectrum', etc.

[image]

When I'm back home in the Netherlands next week I'll give it a try and keep you posted.

AlbrechtL, 2019-08-13:
Great!
I'm closing this issue for now.

closed 2019-08-13 by AlbrechtL

marrem, 2019-08-24:
At home now. In Den Haag, Netherlands. Everything works fine now!

Marc

AlbrechtL, 2019-08-26:
Thanks for the feedback!