AlbrechtL/welle.io #382: Version History

Observerist, 2019-07-31:
Is there somewhere a version history / change log for welle.io ? 

I'm still on the very old 1.0-rc1, built for my macbook with macports.
Before letting myself endure the pain of rebuilding the newest version (2.x ??), I just wanted to see what I'm missing with my old version and if an update is worth the hassle.

BTW ... once build, can I transfer that build with it's dependencies to another mac without having to rebuild everything ( and download a massive amount of collateral apps like xcode again ) ?

Anyone here with macOS experience ?

Observerist, 2019-07-31:
Ooops, sorry, found the release page.

Saw that the v2 betas have a GUI redesign. 
Any screenshots to see the new look ?
Or is this just redesigned under the hood ?

Still, my question about some kind of "portability" of the macOS build remains ...

andimik, 2019-08-07:
A search engine will find screenshots as well...

See https://www.rtl-sdr.com/welle-io-dab-dabdecoder-version-2-released/

Observerist, 2019-08-08:
Cool, Danke für den Link.
    
    Gerade auf der Seite, die mich vor Jahren zur SDR Welt hinzog,
    befand sich die Mitteilung zur v2 und ich Dussel hab sie übersehen.
    
    Die beiden Screenshots sind genau das, was ich suchte.
    
    
    Wenn es jetzt nur leichter wäre, welle.io auf Macs zu installieren.
    
    Ich weiß schon, kein Mac vorhanden, aber ich darf ja noch hoffen,
    oder.
    
    
    Schönen Gruß vom Pariser ORF Korrespondenten Büro,
    
    alex
    
    
    
    
    On 07/08/2019 07:22, andimik wrote:
    
    
      
      A search engine will find screenshots as well...
      See https://www.rtl-sdr.com/welle-io-dab-dabdecoder-version-2-released/
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/AlbrechtL/welle.io/issues/382?email_source=notifications\u0026email_token=AK5WA7XZZXOIDCW3IUMKFFDQDJLZFA5CNFSM4IIETIRKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3XHO5Q#issuecomment-518944630",
"url": "https://github.com/AlbrechtL/welle.io/issues/382?email_source=notifications\u0026email_token=AK5WA7XZZXOIDCW3IUMKFFDQDJLZFA5CNFSM4IIETIRKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3XHO5Q#issuecomment-518944630",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Observerist, 2019-08-09:
Updated to version 2.0 on my Mac, via Macports and Qt. 
As painful as the first build, at least for a non-dev like me, but the result was worth the hassle.

Great GUI, full of smart display choices. And the beta is really stable.
Although getting modules to appear, I have to choose the ones I want to see and restart the app. Otherwise these modules stay invisible, somehow outside of the main frame, on the right edge, where I can see 2 letters from the display name. Even expanding the window to Full HD doesn't bring them in.
But once restarted, several displays are sharing the main space and it's a great ride !!!

Handling Favorites is a blast, simple and intuitive. The Mute button is very helpful, even if this has not become a volume slider. For now, I'm using Audio-Hijack to lower the base output level. Without this leveling,  headphones can't go low enough before muting. 
But for those feeble internal loudspeakers on a MacbookPro, it's just low enough.

The CPU load is lower, I can now stay in the expert mode without losing samples.

One remark about the design of the signal strength bars : 
Dark green bars on dark grey bars is not quite readable.
The bars could get a lighter gray or just be outlined with no grey fill at all.

Having to work/edit every day on Macs with dark GUIs, it's a relief to see some bright backgrounds coming back. Kudos for that choice. But I guess, some people may want a switchable display mode.

All in all, it was really worth the difficult update.

Thank you !

AlbrechtL, 2019-08-10:
Thanks for your feedback!

BTW: The version history can be seen in detail here https://github.com/AlbrechtL/welle.io/commits/master. But I guess this is only readable for experts ;-)

Observerist, 2019-08-10:
:-)) 
Not that kind of expert, indeed.
I'm only good at shooting and editing my camera work. 

But still glad to have access to so many brilliant creations, especially if they come from another domain as mine.
Thx for letting me enjoy the good stuff !!!

closed 2019-08-11 by AlbrechtL