AlbrechtL/welle.io #383: welle.io can not find any DAB+ stations

Rootinator, 2019-08-01:
Hello, 

I would like to use welle.io on the Raspberry with a RTL2832U stick, but unfortunately no DAB+ stations are found. 

If I connect the stick on the smartphone, on which the app welle.io is installed, numerous transmitters are found.  

How can I fix the error?

jay-to-the-dee, 2019-08-06:
🤔 Hmmm are you compiling welle.io yourself,  or downloading pre-compiled from website?

andimik, 2019-08-07:
Have you blacklisted the stick?

AlbrechtL, 2019-08-10:
To help you we need more information.
Which version do you are using? 
Please post also the command line output of welle.io.

Rootinator, 2019-08-30:
I use software that integrates welle.io (Viktorgino head unit)

AlbrechtL, 2019-09-06:
Which version of welle.io is it?

AlbrechtL, 2019-12-02:
Closed because of no activity.

closed 2019-12-02 by AlbrechtL