AlbrechtL/welle.io #384: dab-concurrent: buffer full when using welle-cli

haditim, 2019-08-09:
Hi,
I use welle-cli for its API. However, after a while I get the `dab-concurrent: buffer full`. I use the following command
``welle-cli -c 7D -C 1 -w 1234``
I also tried adjusting `-C` value to 3, 5 etc. as well as doing `-Dw` with no difference. Do you have any idea what could cause this?
Thanks

haditim, 2019-08-09:
I forgot to mention that I grab 3 or 4 mp3 programs on the client side and dump the files.

mpbraendli, 2019-08-12:
Hi @haditim 

do you ensure that you always consume the mp3 data, and never stall the HTTP connection?

Also, how long is "after a while"?

haditim, 2019-08-12:
Hi @mpbraendli 
Thanks for the reply. I think I was stalling the HTTP connection. Now approaching it another way, it works just fine.

closed 2019-08-12 by haditim