AlbrechtL/welle.io #387: Specify QML style Controls.2 to avoid display issues in unsupported styles

tenzap, 2019-08-14:
Hello

I'm running welle.io on debian with the debian package and on kde/plasma
version 2.0-beta2

It turns out that when an application using QtQuickControls2 is run on kde it uses the specific QML style designed for KDE (package qml-module-org-kde-qqc2desktopstyle from source package qqc2-desktop-style which is part of KDE), and it doesn't play well with welle.io (at least with version 5.54 of kde). Issue is described here 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=934749

By searching around I found that there are not many themes which fit well with welle.io and therefore could you fix the style to choose in the configuration of welle.io ?
There are some hints here on how to do that https://doc.qt.io/qt-5/qtquickcontrols2-configuration.html

The styles that work best are :
- Universal
- designer

With :
- Material
- Imagine
- Fusion 
there are some issues (with the dropdown list, or colors of text/icons 
depending on background)

With "org.kde.desktop" there are even more issues (that I described in the bug report in debian)

My current choice is running it with Universal (which can also be themed) by setting this environment variable
`QT_QUICK_CONTROLS_STYLE=Universal welle-io`

Thanks for your great app !

tenzap, 2019-08-21:
Here is the kde bug report since this is still present in the latest versions of qml-module-org-kde-qqc2desktopstyle (5.60 & 5.61) : https://bugs.kde.org/show_bug.cgi?id=411127

AlbrechtL, 2019-08-24:
Thanks for figuring this out! In the past I had some bug reports from users they reported that welle.io is not running on KDE. Now I know why - Thanks!

I don't quite understand what the real cause for that is. Is it a KDE bug, a Qt bug or a welle.io bug?

It can't be a general think because I know from users the welle.io runs on KDE well (as far as is remember is was OpenSuSE).

tenzap, 2019-09-04:
For me the issues comes because some elements you use for the GUI are not supported/implemented in the QtQuick style used by KDE/Plasma (and in a more moderate proportion in Material, Imagine & Fusion styles)

I just checked with OpenSuse Tumbleweed Live CD dated 2019-09-02 and I have the same problem as on debian buster: ie. if one runs plasma/kde then by default the qqc2 style that is used will be _org.kde.desktop_ which renders the app hardly usable.

For completeness of the report here are the steps to reproduce (base on the AppImage because I didn't want to recompile completely the app on a livecd):
Load the live DVD in a virtualmachine, download the AppImage and extract it (with option --appimage-extract). Then I modified the qt.conf file in squashfs-root/usr/bin so that prefix is `prefix = /usr/lib64/qt5`

Finally I executed the following commands in squashfs-root/usr/bin

`LD_LIBRARY_PATH=/usr/lib64 QT_QUICK_CONTROLS_STYLE=Universal ./welle-io`
-> Display is fine

`LD_LIBRARY_PATH=/usr/lib64 QT_QUICK_CONTROLS_STYLE=org.kde.desktop ./welle-io`
or
`LD_LIBRARY_PATH=/usr/lib64 ./welle-io`
-> icons & widget not displayed

Since the all the widgets seems well displayed only with Universal or designer style, that's why I suggested in the initial post you explicitly define the Qt Quick Controls Style that work well directly in the source code.

tenzap, 2019-09-05:
Actually "designer" style doesn't seem to exist. 
But style named "Default" does exist and works well too. Maybe that's even a better choice that Universal. https://doc.qt.io/QT-5/qtquickcontrols2-styles.html#available-styles

mpbraendli, 2019-09-09:
Thanks a lot for this report, that explains why welle.io never looked right on my KDE machine! (On ArchLinux)

To set style in C++, I had to modify the CMakeLists and the qmake pro files to add QuickControls2:

    find_package(Qt5 COMPONENTS Widgets Quick QuickControls2 Multimedia Charts Qml REQUIRED)

and add `Qt5::QuickControls2` to `target_link_libraries`.

`.pro` files:

    QT += quickcontrols2 network qml quick charts multimedia

And in `main.cpp`:

    #include <QQuickStyle>

    QQuickStyle::setStyle("Default");

I set the style right after setting the window icon.

@AlbrechtL shall I push these changes? You can of course change the preferred style if you want.

AlbrechtL, 2019-09-10:
@mpbraendli Thanks for finding a way to fix it in code. I don't know if `QQuickStyle::setStyle("Default");` has some side effects but we will see it :-). Please add it.

mpbraendli, 2019-09-13:
It's in next, in commit 0583c30

tenzap, 2019-09-14:
Thank you ! Display is fine on kde now :)

However the QT_QUICK_CONTROLS_STYLE env.var. is not taken into account anymore (if one wants to change the style, for example when someone is willing to test/fix the "faulty" styles (like org.kde.desktop)...).

tenzap, 2019-09-22:
I made a few changes to some QML files to fix all the GUI issues with all the styles including KDE. As a consequence, it is not needed anymore to hardcode the style.
- I chose that the title of the views in the expert view (like "Service overview", "MOT Slide show") would follow the Style (until now the title was fixed on a white background), but if you prefer to keep  it the same as before, it is possible to have it black over white background whatever the style
- with KDE it was not possible to get the icons colored when they are in a button, that's the only thing that I couldn't achieve
- This also fixes #398 
- If you like the result, I can make a PR
- Finally, if you could rename the icons folder from icon -> icons that would be great because for an obscure reason with kde it does find the icons when placed in "icons" but not in "icon". Do you want a new Issue for that ?

**Below are the results for each style**
* Default
[image]
* Imagine
[image]
* KDE
[image]
* Material
[image]
* Fusion
[image]
* Universal
[image]

tenzap, 2019-09-26:
@mpbraendli, your commit 0583c30 also sets 
`m_device = SoapySDR::Device::make("driver=uhd");`
I guess this is not expected

mpbraendli, 2019-09-26:
Ah dammit! Thanks for mentioning this! Reverted in 532fe91

AlbrechtL, 2019-11-13:
Merged in https://github.com/AlbrechtL/welle.io/commit/1ed44a7cc233d9b13659f85427eb4f9109b732cb.

closed 2019-11-13 by AlbrechtL