AlbrechtL/welle.io #388: New french translation

tenzap, 2019-08-15:
Hello,

Please fin attached a new french translation for 2.0-beta2

[fr_FR.tar.gz](https://github.com/AlbrechtL/welle.io/files/3505208/fr_FR.tar.gz)

Could you also update the GUI to add that translation in the dropdown list of the settings ?

Thanks

tenzap, 2019-08-15:
Please consider this new version which carries the standardized french translations of the programme type codes according to the "ETSI TS 101 756 V2.2.1 (2017-08) (page 23)"

[fr_FR-v2.tar.gz](https://github.com/AlbrechtL/welle.io/files/3505387/fr_FR-v2.tar.gz)

AlbrechtL, 2019-08-24:
Thanks for the translation. Please open a pull request for it.

tenzap, 2019-08-24:
I can provide a patch if you prefer, but no pull request sorry.

AlbrechtL, 2019-08-24:
I really would like to add your translation into welle.io.

The process to do it is via pull requests. They are easy to create and easy to handle. For more information about pull request read the [guide](https://help.github.com/en/articles/about-pull-requests), please.

tenzap, 2019-09-06:
It seems one needs both .ts & .qm
[fr_FRwithQm.tar.gz](https://github.com/AlbrechtL/welle.io/files/3583930/fr_FRwithQm.tar.gz)

AlbrechtL, 2019-09-06:
Please use a pull request to send your patch.

tenzap, 2019-09-09:
here you have. I'm happy it was possible to do this only with the browser
https://github.com/tenzap/welle.io/pull/1

AlbrechtL, 2019-09-10:
Great thanks!
You have to make the pull request on this repository not your fork. Please try it again :-)

tenzap, 2019-09-11:
New try here: https://github.com/AlbrechtL/welle.io/pull/400

closed 2019-10-01 by AlbrechtL