AlbrechtL/welle.io #390: [airspy] AGC seems to be not working

AlbrechtL, 2019-08-24:
In my setup I need the maximum gain to receive a signal. The airspy AGC doesn't increase the gain in my case, so I need the use the manual  gain.

AlbrechtL, 2019-09-07:
I did something in https://github.com/AlbrechtL/welle.io/commit/66832e2ccf968344c2575c5e4bbb77813deb73d4. Now the AGC does something but there is still something to do.

@mpbraendli  How did you found the value 200? I changed it to 10 to make the AGC faster.
https://github.com/AlbrechtL/welle.io/blob/66832e2ccf968344c2575c5e4bbb77813deb73d4/src/input/airspy_sdr.cpp#L204

Is the value 0.2 found by experiments or is some formula behind?
https://github.com/AlbrechtL/welle.io/blob/66832e2ccf968344c2575c5e4bbb77813deb73d4/src/input/airspy_sdr.cpp#L208

mpbraendli, 2019-09-09:
On my airspy unit setting maximum gain manually is not optimal, because the constellation becomes more fuzzy. A bit above mid-range is where the optimum is.

Those values were experimentally found, so you can also try different values to see if it is better.

dwarning, 2022-04-08:
I can confirm the problem with my airspy mini. I get about 8 .. 10dB lower SNR compared to qt-dab. This is for manual gain setting too! Because libairspy functions for gain settings are used I think the problem must have another root cause.
I am worried about the comment inside airspy_sdr.cpp that the sample rate for airspy is 4096ksps. As I know the mini supports standard 3 and 6Msps and 10Msps in a special mode. Airspy-r2 2.5 and 10 MSps.
`dw@modsys:~$ airspy_info
airspy_lib_version: 1.0.11

Found AirSpy board 1
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-6-g4008185 2020-05-08
Part ID Number: 0x6906002B 0x00000030
Serial Number: 0xXXX
Supported sample rates:
	6.000000 MSPS
	3.000000 MSPS
Close board 1
`
Perhaps the decimation by 2 follows in wrong amplitudes?

AlbrechtL, 2022-08-21:
@dwarning If you like you can check qt-dabs code to find differences. PRs are welcome!

AlbrechtL, 2026-10-22:
Added in https://github.com/AlbrechtL/welle.io/commit/863c142d30fd3a0301dd314b4dcb1dc39f356b25.

closed 2026-10-22 by AlbrechtL