AlbrechtL/welle.io #391: RTL_TCP address format

jpuigs, 2019-08-29:
Is is possible to modify the way we set the RTL-TCP server ?
If I'm away from home, and I want to use my home server, I use a ddns.net server, so it would be good to be able to enter an address like home.mydnsserver.com or an IP, both ways.

AlbrechtL, 2019-09-06:
Interesting idea. I'm not sure if it makes sense to transfer I/Q RAW data via internet but I agree to have also the possibility to add a host name instead of a IP address.

jpuigs, 2019-09-08:
It makes sense, because if you are away from home, and want to receive a home server......

I thought that although it's currently not possible to add a hostname into configuration menu, I could do it on Windows registry…..   but I'm not able to….., because IP is stored using bytes, not as a string….

[image]

iuliandj, 2019-12-05:
Hello guys,
Any info about this situation ?
I have also a home server that use rtl_tcp for DAB streaming and my server IP address is dynamic. For remote connection it would be good to connect via DDNS host instead of IP. 
AlbrechtL , on 2.1 vers. is still IP. 
Can you modify to connect by DDNS host please ? 
Than you !

AlbrechtL, 2019-12-08:
Added in https://github.com/AlbrechtL/welle.io/commit/af7405cd3fd0c26b60940a4d994b398d6a108592.

[image]

Please test the latest [nightly version](https://bintray.com/albrechtl/welle.io/welle.io_nightly#files) or build it from the next branch.

closed 2019-12-08 by AlbrechtL

iuliandj, 2019-12-11:
It works ! Thank you !