AlbrechtL/welle.io #392: Error completing welle.io / QT

Rootinator, 2019-08-30:
Hello,

I have trouble completing welle.io (DAB Software).
I can not install QT.

From "Step 7" you have to execute the commands on the host computer, in my case a Raspbian running in (Windows) VMware.

I come up to and including "Step 11," but from "Step 12," it says the command "-device-option" was not found.

[code]
pi@raspberry:~/raspi/qt5 $ -device-option CROSS_COMPILE=~/raspi/gcc-linaro-5.5.0-2017.10-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-
-bash: -device-option: command not found
[/code]

Thus also "make" and "make install" of QT does not work.

[Code]
pi@raspberry:~/raspi/qt5 $ make
make: *** No targets specified and no makefile found.  Stop.
[/code]

I inserted the commands via Copy & Paste in the SSH console of the host computer.

Obviously there was no error downloading QT ...

Where is the mistake?

(https://github.com/AlbrechtL/welle.io/blob/master/RASPBERRY-PI.md)

Rootinator, 2019-09-06:
Is there no solution for that?

@AlbrechtL

AlbrechtL, 2019-09-06:
The Raspberry Pi how to is outdated see https://github.com/AlbrechtL/welle.io/issues/338.
You also have `qmake` instead of `make`.

Rootinator, 2019-09-07:
I use

`pi@raspberry:~/raspi/qt5 $ ./configure -release -opengl es2 -device-option CROSS_COMPILE=~/raspi/gcc-linaro-5.5.0-2017.10-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-`

and get the following message:

`+ cd qtbase
+ /home/pi/raspi/qt5/qtbase/configure -top-level -release -opengl es2 -device-option CROSS_COMPILE=/home/pi/raspi/gcc-linaro-5.5.0-2017.10-i686_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-
<srcbase> = /home/pi/raspi/qt5/qtbase
<outbase> = /home/pi/raspi/qt5/qtbase
Creating qmake...
.Done.
Info: creating stash file /home/pi/raspi/qt5/.qmake.stash

Selecting Qt Edition.

Type 'c' if you want to use the Commercial Edition.
Type 'o' if you want to use the Open Source Edition.

Which edition of Qt do you want to use? o

This is the Qt Open Source Edition.

You are licensed to use this software under the terms of
the GNU Lesser General Public License (LGPL) version 3
or the GNU General Public License (GPL) version 2.

Type 'L' to view the GNU Lesser General Public License version 3 (LGPLv3).
Type 'G' to view the GNU General Public License version 2 (GPLv2).
Type 'y' to accept this license offer.
Type 'n' to decline this license offer.

Do you accept the terms of either license? y

Running configuration tests...
Checking for gold linker... yes
Checking for machine tuple... yes
Checking for valid makespec... yes
Checking for target architecture... i386
Checking for SSE2 instructions... yes
Checking for AES new instructions... yes
Checking for alloca() in alloca.h... yes
Checking for SSE3 instructions... yes
Checking for SSSE3 instructions... yes
Checking for SSE4.1 instructions... yes
Checking for SSE4.2 instructions... yes
Checking for AVX instructions... yes
Checking for AVX2 instructions... yes
Checking for AVX512 F instructions... yes
Checking for AVX512 BW instructions... yes
Checking for AVX512 CD instructions... yes
Checking for AVX512 DQ instructions... yes
Checking for AVX512 ER instructions... yes
Checking for AVX512 IFMA instructions... yes
Checking for AVX512 PF instructions... yes
Checking for AVX512 VBMI instructions... yes
Checking for AVX512 VL instructions... yes
Checking for C++14 support... yes
Checking for C++1z support... yes
Checking for pkg-config... yes
Checking for D-Bus >= 1.2... yes
Checking for new dtags support... yes
Checking for F16C instructions... yes
Checking for D-Bus >= 1.2 (host)... yes
Checking for udev... no
Checking for POSIX fallocate()... yes
Checking for precompiled header support... yes
Checking for RDRAND instruction... yes
Checking for symbol visibility support... yes
Checking for -Bsymbolic-functions support... yes
[... 287 lines cut ...]
Note: Dropped compiler flags '-pthread' when detecting library 'glib'.

Note: Disabling X11 Accessibility Bridge: D-Bus or AT-SPI is missing.

Note: No wayland-egl support detected. Cross-toolkit compatibility disabled.

ERROR: Feature 'opengles2' was enabled, but the pre-condition 'config.win32 || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.

ERROR: The OpenGL functionality tests failed!
You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

Check config.log for details.`

qmake or make are not found:

`pi@raspberry:~/raspi/qt5 $ qmake
-bash: qmake: command not found`

`pi@raspberry:~/raspi/qt5 $ make
make: *** No targets specified and no makefile found.  Stop.`

Why does not that work?

I used the "device-option" for 32 bits, because the host system has 32 bits.

AlbrechtL, 2019-09-10:
You have some errors during running `configure`. For this case ask the Qt community please.

closed 2019-09-10 by AlbrechtL