AlbrechtL/welle.io #393: dabdate command line tool

alexmyczko, 2019-09-04:
Hi

There's a tool dvbdate from dvb-apps (Switzerland stopped DVB-T and does not provide DVB-T2), how hard would it be to create dabdate printing date+time gotten via rtlsdr device?

There are a few people that'd like to get the time from the air, without internet. I'm one of them.

Best,

AlbrechtL, 2019-09-06:
Interesting idea to use DAB to get a date and time. What is the use case to use DAB or DVB rather than DCF77?

andimik, 2019-09-06:
Indeed, it's an interesting idea.

But you get the exact time when you plug in a GPS mouse (some models are even supported by the kernel).

alexmyczko, 2019-09-06:
77.5 khz is not the cheap rtlsdr range...

mpbraendli, 2019-09-24:
When you use welle-cli in webserver-mode, you can retrieve the JSON data, which includes UTC time.

alexmyczko, 2019-10-01:
> When you use welle-cli in webserver-mode, you can retrieve the JSON data, which includes UTC time.

@mpbraendli any chance of creating a simple shell script example for inclusion and usage?

mpbraendli, 2019-10-01:
Very low chance, as I don't need this myself. Try combining wget or curl to fetch the `mux.json` with [jq](https://stedolan.github.io/jq/) to filter out the UTC time.

alexmyczko, 2019-10-03:
Okay this is what I came up with: 

`printf "%02d %02d %04d %02d:%02d:$(date +%S) UTC\n" $(curl -s http://localhost:7979/mux.json |gron |grep utctime |gron -u | python3 -c "import json,sys;obj=json.load(sys.stdin);print (obj['utctime']['day'], obj['utctime']['month'], obj['utctime']['year'], str(obj['utctime']['hour'])+' '+str(obj['utctime']['minutes']));")`

gvanem, 2019-10-03:
What does `gron` do?

closed 2019-11-28 by AlbrechtL