AlbrechtL/welle.io #394: Depending on QtQuick style, content of ConsoleOutput View is empty/blank

tenzap, 2019-09-05:
To reproduce, launch welle-io by defining a specific qtQuick style. (This is on linux)

It works fine with style
QT_QUICK_CONTROLS_STYLE=Default welle-io
QT_QUICK_CONTROLS_STYLE=Material welle-io
QT_QUICK_CONTROLS_STYLE=Fusion welle-io

It is empty for example with style :
QT_QUICK_CONTROLS_STYLE=Universal welle-io
QT_QUICK_CONTROLS_STYLE=Imagine welle-io

Until now, the only style for which I haven't had any problems with is "Default". All other induce at least one issue in the GUI, some even more

AlbrechtL, 2019-09-06:
Thanks for reporting.
It looks like this is related to your previous bug report https://github.com/AlbrechtL/welle.io/issues/387.

AlbrechtL, 2019-10-02:
Is this issue solved with the changed discussed in #387?
If yes I would like to close it and to add the tag duplicated.

tenzap, 2019-10-02:
it is. I thought I mentionned it.

AlbrechtL, 2019-10-02:
Closed because this issue is handled in #387.

closed 2019-10-02 by AlbrechtL