AlbrechtL/welle.io #396: [Android] Station scan stucks on 5A

AlbrechtL, 2019-09-07:
Reported in https://github.com/AlbrechtL/welle.io/issues/373. Thanks to @Mikescotland!

MichalLeonBorsuk, 2019-11-19:
Hi, 

I'm a user, not one of the developers. 

As far as I have found, this happens when 1) automatic gain is on (the default), 2) manual gain had been set to a very high value, which somehow confuses the receiver into ignoring AGC.

What I think happens is that welle.io thinks the signal is strong enough and waits for the phase synch. 

Workaround:
1. Disable AGC
2. Set gain to a small value, e.g. 14
3. Enable AGC
4. If still no go, then repeat with a smaller value. 

Devices tested: two different RTL-SDR.com sticks

AlbrechtL, 2019-12-04:
Fixed in https://github.com/AlbrechtL/welle.io/commit/f397ff0a28fde933965231657766d6040a341b56.

closed 2019-12-04 by AlbrechtL