AlbrechtL/welle.io #399: Freeze when chosing RAW file

andimik, 2019-09-09:
I'm using Kubuntu 64bit 18.04 and tried under various window managers (KDE, Cinnamon), but when I select "RAW file" welle.io hangs and I cannot load any file.

[image]

I even have the same issue with older AppImages (20190303_477d558_Linux_welle-io-x86_64.AppImage and 20190505_9f1180f_Linux_welle-io-x86_64.AppImage on my laptop). No matter if I switch to English languague or use the full screen mode.

```
$ ./apps/welle.io/src/welle-gui/welle-io -v
QML debugging is enabled. Only use this in a safe environment.
welle.io 2.0-Beta 3 Git: 65a09ee
```

AlbrechtL, 2019-09-10:
Thanks for reporting.
Please post the command line output.

andimik, 2019-09-17:
Well, finally I found the root cause and could solve it myself, but it was a bit tricky to understand why this happened:

```
$ gdb /home/andreas/apps/welle.io/src/welle-gui/welle-io
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/andreas/apps/welle.io/src/welle-gui/welle-io...done.
(gdb) r
Starting program: /home/andreas/apps/welle.io/src/welle-gui/welle-io 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
QML debugging is enabled. Only use this in a safe environment.
[New Thread 0x7fffea3c0700 (LWP 10119)]
[New Thread 0x7fffdf5b4700 (LWP 10120)]
[New Thread 0x7fffdedb3700 (LWP 10121)]
[New Thread 0x7fffdccac700 (LWP 10122)]
[New Thread 0x7fffcffff700 (LWP 10123)]
[New Thread 0x7fffcf7fe700 (LWP 10124)]
2019-09-17T22:16:10.587 Debug: Audio: ActiveState
[New Thread 0x7fffaff51700 (LWP 10125)]
[New Thread 0x7fffaefe7700 (LWP 10126)]
[New Thread 0x7fffae7e6700 (LWP 10127)]
2019-09-17T22:16:14.970 Debug: main: Set language "en_US"
2019-09-17T22:16:14.970 Debug: main: Error while loading language "en_US" use English "en_GB" instead
2019-09-17T22:16:15.006 Debug: main: Set language "de_DE"
2019-09-17T22:16:15.018 Debug: Apply settings initially
2019-09-17T22:16:15.018 Debug: RadioController: Close device
2019-09-17T22:16:15.021 Debug: RadioController: AGC off
2019-09-17T22:16:15.021 Debug: Audio: Volume 1
2019-09-17T22:16:15.172 Warning: file:///home/andreas/Qt/5.11.1/gcc_64/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window'
2019-09-17T22:16:15.220 Debug: os: linux
2019-09-17T22:16:15.220 Debug: desktopAvailableWidth: 1366
2019-09-17T22:16:15.220 Debug: desktopAvailableHeight: 742
2019-09-17T22:16:15.220 Debug: orientation: 2
2019-09-17T22:16:15.220 Debug: devicePixelRatio: 1
2019-09-17T22:16:15.220 Debug: pixelDensity: 3.9751024219785513
2019-09-17T22:16:15.220 Debug: Creating component: qrc:/QML/RadioView.qml
2019-09-17T22:16:15.249 Debug: Creating component: qrc:/QML/MotView.qml
2019-09-17T22:16:15.255 Debug: Creating component: qrc:/QML/expertviews/TextOutputView.qml
2019-09-17T22:16:15.284 Warning: QGridLayoutEngine::addItem: Cell (0, 0) already taken
2019-09-17T22:16:22.901 Debug: RadioController: Close device
```

The last line was the click on RAW file, but then I saw that in the console (thanks to gdb)

```
^C
Thread 1 "welle-io" received signal SIGINT, Interrupt.
0x00007ffff35aad19 in __libc_open64 (file=0x555556240e70 "/home/andreas/stdin.raw", file@entry=0x8 <error: Cannot access memory at address 0x8>, 
    oflag=0) at ../sysdeps/unix/sysv/linux/open64.c:47
47	../sysdeps/unix/sysv/linux/open64.c: No such file or directory.
```

In the hidden config file `~/.config/welle.io/welle.io.conf` there was 

`rawFile=/home/andreas/stdin.raw`

but this was just a fifo file in my case I used last time.

I've deleted this line and now welle-io does not freeze any more.

closed 2019-09-17 by andimik