AlbrechtL/welle.io #402: RTL_SDR:Setting gain failed

tenzap, 2019-09-15:
In the log I often have these errors:
```

rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=05 len=1
2019-09-15T21:46:09.766 Info: RTL_SDR:Setting gain failed
```

By looking around, if try to modify the code so that it calls the faulty "rtlsdr_set_tuner_gain" (from rtl-sdr.cpp) a second time, it seems to work.
Could this be related to the fact that this is called while we are in a rtlsdr_read_async
(like the same error in the case we call rtlsdr_set_center_freq while not having cancelled the rtlsdr_read_async)

This message sometimes appears many times in the log, even when only playing a station.
It also appears when scanning

tenzap, 2019-09-16:
This seems to happen only if the active antenna is on and amplifies the signal

AlbrechtL, 2019-10-01:
I also saw it sometimes at the logs from other users. In my case I never experienced it.

What you can do is just looking to the source code of other rtl-sdr open source programs. Maybe they did something different than we do.

AlbrechtL, 2019-12-02:
The messages `r82xx_write: i2c wr failed=-9 reg=05 len=1` are looking to me for a hardware issue.

closed 2019-12-02 by AlbrechtL