AlbrechtL/welle.io #403: `ofdm-processor: SyncOnNull failed` indefinitely at end of channel scan

tenzap, 2019-09-16:
After the end of scanning, the console log displays `ofdm-processor: SyncOnNull failed` indefinitely (until one start eg playing a station)

```
2019-09-16T09:14:28.626 Info: ofdm-processor: SyncOnNull failed
2019-09-16T09:14:28.722 Info: ofdm-processor: SyncOnNull failed
2019-09-16T09:14:28.722 Debug: RadioController: Stop channel scan
2019-09-16T09:14:28.818 Info: ofdm-processor: SyncOnNull failed
2019-09-16T09:14:28.914 Info: ofdm-processor: SyncOnNull failed
2019-09-16T09:14:29.010 Info: ofdm-processor: SyncOnNull failed
2019-09-16T09:14:29.106 Info: ofdm-processor: SyncOnNull failed
2019-09-16T09:14:29.202 Info: ofdm-processor: SyncOnNull failed
...
```

AlbrechtL, 2019-10-01:
Thanks for reporting!

AlbrechtL, 2019-12-09:
Fixed in https://github.com/AlbrechtL/welle.io/commit/94c7b3bec4e4387069edeaf43cf4ac689c0f2d2b.

closed 2019-12-09 by AlbrechtL

heydojo, 2019-12-15:
@AlbrechtL
This isn't fixed for me. I see the log spammed with:
`Info: ofdm-processor: SyncOnNull failed`
And there is no sound.

This is when using the snapshot **20191211_ce49a23_Windows_welle-io-setup.exe** from https://bintray.com/albrechtl/welle.io/welle.io_nightly#files which contains the fix mentioned in 94c7b3bec4e4387069edeaf43cf4ac689c0f2d2b.

AlbrechtL, 2019-12-27:
>  `Info: ofdm-processor: SyncOnNull failed`
> And there is no sound.

The `SyncOnNull failed` message means that welle.io can't find a signal on the chosen channel.