AlbrechtL/welle.io #404: Restart play after suspend/standby

tenzap, 2019-09-19:
On Linux, after I suspend/standby the PC, welle io says RTL-SDR is unplugged.
Actually I didn't unplug it and the only way I found for now to get it playing again is to close/reopen the software.

Could it be possible to manage that case and continue playing when then system comes back from standby ?

AlbrechtL, 2019-10-01:
Your observations makes sense. The system removes all USB devices and welle.io is still running.

The only options that we have is to open the rtl-sdr again after it is unplugged. At the moment this happens only when start welle.io or when trying changing the channel (if I remember it correctly).

This feature could be implemented in the future.

tenzap, 2021-02-27:
On linux, this should work once MPRIS PR #543 is merged.

It is because on linux (at least KDE) MPRIS detects the system is going to standby and stops playback of any audio app supporting MPRIS. Then, thanks to the start/stop feature, clicking play will be sufficient to restart playing.

Without MPRIS support, it doesn't work on linux probably because Welle-io on its own doesn't stop playback on that standby event.

AlbrechtL, 2021-05-22:
Thanks for the PR. I'm closing this issue because we have a solution that fits for 99% of the Linux users.

closed 2021-05-22 by AlbrechtL