AlbrechtL/welle.io #406: Command line options are not translated

tenzap, 2019-09-19:
running `welle-io -h` shows that the command line options are not translated.

Adding this right after the `QApplication app(argc, argv);` in `main.cpp` fixes it

```
    //Initialise translation
    QTranslator *translator = new QTranslator;
    QString Language = QLocale::system().name();
    qDebug() << "main:" <<  "Set language" << Language;
    bool isTranslation = translator->load(QString(":/i18n/") + Language);
    if(!isTranslation && Language != "en_GB")
    {
        qDebug() << "main:" <<  "Error while loading language" << Language << "use English \"en_GB\" instead";
        Language = "en_GB";
    }
    QCoreApplication::installTranslator(translator);
```

But this is duplication of your code in gui_helper.cpp :/

This is on linux

tenzap, 2019-09-19:
This also fixes the untranslated actions when right clicking on the system tray icon

tenzap, 2019-09-28:
I have a cleaner fix for that, but prefer to wait your decision on the other ones because there could be conflicts

AlbrechtL, 2019-10-01:
Thanks for reporting!

Please open an new PR for this because it makes sense for the version 2.0 final. At the moment I go over all your PRs. If some PRs are in conflict state we can solve it afterwards. First I want to get out version 2.0 and thus I only merge bug fixes. New features can be merged after the version 2.0 release.

tenzap, 2019-10-01:
See #430 

Can I suggest you to accept also these PR for 2.0 because none of these messages are translatable in the GUI : #407 #408 #409 Or do you prefer me to open an issue for each PR ? I didn't do it initially to avoid duplicating everything

AlbrechtL, 2019-10-02:
I looked over your tanslatation releated pull request. Because of the full bunch of changes I decided to propose the merge to version 2.1 (https://github.com/AlbrechtL/welle.io/projects/2).

tenzap, 2019-11-15:
fixed by #430

closed 2019-11-15 by tenzap