AlbrechtL/welle.io #416: Warning: Icon theme "welle_io_icons" not found.

tenzap, 2019-09-23:
When on KDE with "org.kde.desktop" style there is this message in the console and icons are not displayed;
`Warning: Icon theme "welle_io_icons" not found.`

There are 2 ways to have this fixed:
- Either copy the icons theme (`welle_io_icons` directory) to a valid system icon directory on linux like "`/usr/share/icons`"
- or rename the base folder of the icons to "icons" instead of "icon" ?

I found this bug in KDE : https://bugs.kde.org/show_bug.cgi?id=405522. Probably this is related.

According to https://sources.debian.org/src/kiconthemes/5.62.0-1/src/kicontheme.cpp/#L289 I understand that in KDE, the system loads preferably the icons in the embedded "/icons" folder. But this directory doesn't exist in welle-io, instead it is called "icon". Hence the 2nd way to fix it.

AlbrechtL, 2019-10-01:
Thanks for figuring this out!

I agree that the situation is not nice at the moment. But all style related changes I will merge after the version 2.0 release.

AlbrechtL, 2019-10-19:
> or rename the base folder of the icons to "icons" instead of "icon" ?

Which base folder do you mean? This one `/src/welle-gui/icon/welle_io_icons`?

tenzap, 2019-10-19:
the `icon` folder in `/src/welle-gui/`

AlbrechtL, 2019-10-22:
In this case I would prefer to rename the folder to `icons`.

tenzap, 2019-10-23:
Ok, please do :)

AlbrechtL, 2019-10-25:
You are welcome to send a pull request.

tenzap, 2019-11-15:
fixed by #449

closed 2019-11-15 by tenzap