AlbrechtL/welle.io #418: empty welle.io window

MAlfare, 2019-09-23:
Hi!
Since an update of my Linux Mint 19.2 Cinnamon today, welle.io window is empty.
I use welle.io 2.0-Beta 3.

Most suspect packages I think, are ones related to mesa.
[mesa-update.txt](https://github.com/AlbrechtL/welle.io/files/3643936/mesa-update.txt)

Regards
Manfred

ontheair81, 2019-09-28:
Having exactly the same problem: After updating Linux Mint 19.2, the GUI of welle.io is empty (but audio still playing with last station). Mesa was updated here too.

Difference: I am using XFCE instead of Cinnamon.
Adding this information: Using welle.io as AppImage. Problem found in welle.io 2.0 Beta 3 and Beta 2.
Confirm: Starting as root makes the GUI work again. Welle.io 1.0 still works without problem, even without root.

andimik, 2019-09-28:
Have you tried to delete the Config file in `~/.config/welle.io`?

ontheair81, 2019-09-28:
Yes, I tried to delete the config file, then starting welle.io. Made no difference, the GUI was still blank.

Right now welle.io works without problems after a first start as root. I have to check if it stays like this after the system is restarted or if welle.io has to be started as root again.

MAlfare, 2019-09-28:
Hi!
For me, it works when started as root only. Starting it without afterwards, it does not.

AlbrechtL, 2019-10-01:
Please post the command line output of the non-working case.

MAlfare, 2019-10-01:
Here it is:
`alf@Ida ~/PROG/welle $ ./welle-io
2019-10-01T14:58:49.416 Debug: Audio: ActiveState
2019-10-01T14:58:50.177 Debug: main: Set language "de_DE"
2019-10-01T14:58:50.219 Debug: Apply settings initially
2019-10-01T14:58:50.219 Debug: RadioController: Close device
2019-10-01T14:58:50.219 Info: InputFactory:Input device:auto
2019-10-01T14:58:50.221 Info: Airspy: Open airspy
2019-10-01T14:58:50.226 Info: Airspy: airpsy_open () failed: AIRSPY_ERROR_NOT_FOUND(-5)
2019-10-01T14:58:50.226 Info: RTL_SDR:Open rtl-sdr
2019-10-01T14:58:50.231 Info: RTL_SDR:Found 1 devices. Uses the first working one
Detached kernel driver
Found Elonics E4000 tuner
2019-10-01T14:58:50.651 Info: RTL_SDR: Opening rtl-sdr device0
2019-10-01T14:58:50.717 Info: RTL_SDR: Supported gain values14
2019-10-01T14:58:50.717 Info: RTL_SDR: gain 42
2019-10-01T14:58:50.717 Info: RTL_SDR: gain 34
2019-10-01T14:58:50.717 Info: RTL_SDR: gain 29
2019-10-01T14:58:50.717 Info: RTL_SDR: gain 24
2019-10-01T14:58:50.717 Info: RTL_SDR: gain 21.5
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 19
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 16.5
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 14
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 11.5
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 9
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 6.5
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 4
2019-10-01T14:58:50.718 Info: RTL_SDR: gain 1.5
2019-10-01T14:58:50.718 Info: RTL_SDR: gain -1
2019-10-01T14:58:50.740 Debug: RadioController: AGC on
2019-10-01T14:58:50.740 Debug: Audio: Volume 1
2019-10-01T14:58:50.750 Debug: RadioController: Play: "Playing last station" "42f3" on channel "9B"
2019-10-01T14:58:50.791 Info: Start RTLSDR thread
2019-10-01T14:58:50.791 Debug: RadioController: Tune to channel "9B" -> 204.64 MHz
2019-10-01T14:58:50.914 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2019-10-01T14:58:50.914 Info: OFDM-processor:start
2019-10-01T14:58:50.924 Info: ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
2019-10-01T14:58:50.993 Info: RTL_SDR: Set bias tee to 0
2019-10-01T14:58:51.087 Warning: file:///tmp/.mount_welle-youksm/usr/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window'
2019-10-01T14:58:51.090 Warning: qrc:/QML/settingpages/GlobalSettings.qml:10:1: QML GlobalSettings: Binding loop detected for property "implicitHeight"
2019-10-01T14:58:51.107 Debug: os: linux
2019-10-01T14:58:51.107 Debug: desktopAvailableWidth: 1280
2019-10-01T14:58:51.107 Debug: desktopAvailableHeight: 997
2019-10-01T14:58:51.107 Debug: orientation: 2
2019-10-01T14:58:51.107 Debug: devicePixelRatio: 1
2019-10-01T14:58:51.107 Debug: pixelDensity: 3.403124337315332
2019-10-01T14:58:51.107 Debug: Creating component: qrc:/QML/RadioView.qml
2019-10-01T14:58:51.235 Debug: Creating component: qrc:/QML/expertviews/ServiceDetails.qml
2019-10-01T14:58:51.286 Debug: Creating component: qrc:/QML/expertviews/SpectrumGraph.qml
2019-10-01T14:58:51.396 Warning: QGridLayoutEngine::addItem: Cell (0, 0) already taken
2019-10-01T14:58:51.396 Debug: Creating component: qrc:/QML/expertviews/ImpulseResponseGraph.qml
2019-10-01T14:58:51.439 Warning: QGridLayoutEngine::addItem: Cell (1, 0) already taken
2019-10-01T14:58:51.440 Debug: Creating component: qrc:/QML/MotView.qml
2019-10-01T14:58:52.723 Info: ofdm-processor: SyncOnPhase failed
2019-10-01T14:58:54.219 Info: ofdm-processor: SyncOnPhase failed
2019-10-01T14:58:54.893 Info: ofdm-processor: SyncOnPhase failed
2019-10-01T14:58:55.465 Debug: RadioController: ID of ensemble: 16898
2019-10-01T14:58:55.467 Info: ofdm-processor: Found sync (coarseCorrector: 0; fineCorrector: 0 after 45 frames)
2019-10-01T14:58:55.641 Debug: RadioController: Found service  4D2B "VIRGIN HITS     "
2019-10-01T14:58:55.679 Debug: RadioController: Found service  4DE2 "ENERGY HITS+    "
2019-10-01T14:58:55.758 Debug: RadioController: Found service  9201 "Radio L+        "
2019-10-01T14:58:55.925 Debug: RadioController: Found service  4DA8 "RADIO MELODY+   "
2019-10-01T14:58:55.958 Debug: RadioController: Found service  4DE3 "ROCKIT RADIO+   "
2019-10-01T14:58:56.008 Debug: RadioController: Found service  4F10 "SUNSHINE Radio +"
2019-10-01T14:58:56.040 Debug: RadioController: Found service  43D3 "COULEUR 3+      "
2019-10-01T14:58:56.278 Debug: RadioController: Found service  4DBC "Freundes-Dienst+"
2019-10-01T14:58:56.408 Debug: RadioController: Found service  4DE4 "VINTAGE RADIO+  "
2019-10-01T14:58:56.440 Debug: RadioController: Found service  42F3 "Swiss Jazz+     "
2019-10-01T14:58:56.479 Debug: RadioController: Found service  4D52 "RADIO GLORIA    "
2019-10-01T14:58:56.525 Debug: RadioController: Found service  43D4 "OPTION MUSIQUE+ "
2019-10-01T14:58:56.558 Debug: RadioController: Found service  4D15 "VIRGIN ROCK     "
2019-10-01T14:58:56.607 Debug: RadioController: Found service  42F2 "Swiss Classic+  "
AACDecoder: using decoder 'FAAD2'
2019-10-01T14:58:56.622 Debug: RadioController: Audio sample rate 48000 Hz, mode= "HE-AAC, 48 kHz Stereo @ 64 kBit/s"
`

AlbrechtL, 2019-10-02:
Weird, everthing look fine to me.

MAlfare, 2019-10-11:
Hi!
After some new updates for different mesa files yesterday, window stays empty, even started with root privileges.
Maybe this warning has to do with the problem?
`2019-10-11T17:32:56.814 Warning: file:///tmp/.mount_welle-juwspS/usr/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window'
`
Attached logs, started welle.io with and without sudo

[sudo_welle.io.log](https://github.com/AlbrechtL/welle.io/files/3718508/sudo_welle.io.log)
[welle.io.log](https://github.com/AlbrechtL/welle.io/files/3718509/welle.io.log)

AlbrechtL, 2019-10-19:
Please post a screenshot.

BTW: What is the meaning of the file `+tmp+.mount_welle-KJflrK+usr+qml+QtQuick+Controls.2+ApplicationWindow.qml.txt`?

MAlfare, 2019-10-19:
Hi!
A screenshot from an empty window?
[image]

Shows the content of:
file:///tmp/.mount_welle-KJflrK/usr/qml/QtQuick/Controls.2/ApplicationWindow.qml
mentioned in the Warning in welle.io.log

AlbrechtL, 2019-10-20:
Oh it is complete empty. I never saw this before. Weird!

But warning `2019-10-11T17:36:26.785 Warning: file:///tmp/.mount_welle-KJflrK/usr/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window' ` is independent from this.

MAlfare, 2019-11-05:
Released Version 2.0 works as expected (and without being started as root)
Thanks for your work.

closed 2019-11-05 by MAlfare