AlbrechtL/welle.io #419: Add possibility to group the stations by Channel

tenzap, 2019-09-24:
On my place there are >80 stations on 7 channels
Depending on the way my antenna is placed I receive some channels or not.
Thus it would be nice to have the ability to group the stations by channel to know immediately the stations I can choose depending on the channels I receive at that time

Display of the station list could be as such:
```
Channel 5A:
   STATION 1
   STATION 2
   STATION 3
Channel 7D:
   STATION A
   STATION B
   STATION C
 ...
```

AlbrechtL, 2019-10-01:
Great idea! Are are welcome to post a GUI mockup how it should look like.

tenzap, 2021-08-22:
related to #487 and #669

closed 2022-01-06 by AlbrechtL

AlbrechtL, 2024-12-28:
FYI: I'm proposing a possible solution here: https://github.com/AlbrechtL/welle.io/issues/487#issuecomment-2563988969