AlbrechtL/welle.io #422: Crash/Abort when closing

tenzap, 2019-09-26:
This is with welle.io 2.0 beta 3
On linux debian
with RTL-SDR

**Steps:**
Play a station, wait a little bit.
After some time, when I close welle.io (through the GUI : (Menu / Exit) ) sometimes there is a crash.
This cannot be reproduced at every close, but happens every now & then

Until now I identified 2 cases :
* `OfdmDecoder::workerthread` with `CRadioController::onConstellationPoints`
* `OFDMProcessor::run` with `CRadioController::onNewImpulseResponse`

Detailed logs are attached : [welle-io kcrash logs.zip](https://github.com/AlbrechtL/welle.io/files/3657050/welle-io.kcrash.logs.zip)
Extraction of the logs is bellow

### `OfdmDecoder::workerthread` with `CRadioController::onConstellationPoints`
```
Thread 14 (Thread 0x7efebe05c700 (LWP 28533)):
[KCrash Handler]
#6  0x00007eff1fb8e7bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#7  0x00007eff1fb79535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#8  0x00007eff1fbd0508 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#9  0x00007eff1fbd6c1a in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#10 0x00007eff1fbd8730 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#11 0x0000564c6033a7e4 in __gnu_cxx::new_allocator<std::complex<float> >::deallocate (this=<synthetic pointer>, __p=<optimized out>) at /usr/include/c++/8/ext/new_allocator.h:116
#12 std::allocator_traits<std::allocator<std::complex<float> > >::deallocate (__a=<synthetic pointer>..., __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/8/bits/alloc_traits.h:462
#13 std::_Vector_base<std::complex<float>, std::allocator<std::complex<float> > >::_M_deallocate (this=<synthetic pointer>, __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:304
#14 std::_Vector_base<std::complex<float>, std::allocator<std::complex<float> > >::~_Vector_base (this=<synthetic pointer>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:285
#15 std::vector<std::complex<float>, std::allocator<std::complex<float> > >::~vector (this=<synthetic pointer>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:570
#16 std::vector<std::complex<float>, std::allocator<std::complex<float> > >::_M_move_assign (__x=..., this=0x7ffe1a6d6d38) at /usr/include/c++/8/bits/stl_vector.h:1679
#17 std::vector<std::complex<float>, std::allocator<std::complex<float> > >::operator= (__x=..., this=0x7ffe1a6d6d38) at /usr/include/c++/8/bits/stl_vector.h:601
#18 CRadioController::onConstellationPoints (this=0x7ffe1a6d6b70, data=...) at ./src/welle-gui/radio_controller.cpp:882
#19 0x0000564c60359b94 in OfdmDecoder::workerthread (this=0x564c6247cca8) at ./src/backend/ofdm-decoder.cpp:120
#20 0x00007eff1ff72b2f in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#21 0x00007eff21b42fa3 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#22 0x00007eff1fc504cf in clone () from /lib/x86_64-linux-gnu/libc.so.6
```

### `OFDMProcessor::run` with `CRadioController::onNewImpulseResponse`
```
Thread 15 (Thread 0x7f57ef7fe700 (LWP 16443)):
[KCrash Handler]
#6  0x00007f584778a7bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#7  0x00007f5847775535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#8  0x00007f58477cc508 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#9  0x00007f58477d2c1a in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#10 0x00007f58477d4730 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#11 0x000055604b18d6f4 in __gnu_cxx::new_allocator<float>::deallocate (this=<synthetic pointer>, __p=<optimized out>) at /usr/include/c++/8/ext/new_allocator.h:116
#12 std::allocator_traits<std::allocator<float> >::deallocate (__a=<synthetic pointer>..., __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/8/bits/alloc_traits.h:462
#13 std::_Vector_base<float, std::allocator<float> >::_M_deallocate (this=<synthetic pointer>, __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:304
#14 std::_Vector_base<float, std::allocator<float> >::~_Vector_base (this=<synthetic pointer>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:285
#15 std::vector<float, std::allocator<float> >::~vector (this=<synthetic pointer>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/stl_vector.h:570
#16 std::vector<float, std::allocator<float> >::_M_move_assign (__x=..., this=0x7ffe40b13bb8) at /usr/include/c++/8/bits/stl_vector.h:1679
#17 std::vector<float, std::allocator<float> >::operator= (__x=..., this=0x7ffe40b13bb8) at /usr/include/c++/8/bits/stl_vector.h:601
#18 CRadioController::onNewImpulseResponse (this=0x7ffe40b13a70, data=...) at ./src/welle-gui/radio_controller.cpp:876
#19 0x000055604b1ae809 in OFDMProcessor::run (this=0x55604d35cca0) at ./src/backend/ofdm-processor.cpp:344
#20 0x00007f5847b6eb2f in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#21 0x00007f584973efa3 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#22 0x00007f584784c4cf in clone () from /lib/x86_64-linux-gnu/libc.so.6
```

AlbrechtL, 2019-10-02:
Thanks for reporting!

AlbrechtL, 2019-11-28:
Does it still happens with the latest next version?

tenzap, 2019-11-28:
Yes it does

AlbrechtL, 2019-12-02:
I can't reproduce it.
1. Does it only happen with `CRadioController::onConstellationPoints `and `CRadioController::onNewImpulseResponse`?
2. Does it also happen without constellation diagram and impulse response view?

tenzap, 2019-12-02:
> Does it only happen with `CRadioController::onConstellationPoints `and `CRadioController::onNewImpulseResponse`?

For now, I only have examples with `CRadioController::onConstellationPoints` and `CRadioController::onNewImpulseResponse`. I couldn't catch a crash on something else

> Does it also happen without constellation diagram and impulse response view?

Yes. It even does happen in standard mode (=non expert mode, = without any expert view)

> I can't reproduce it.

The best moment I found to reproduce it is to quit welle.io between the moment you start playing a station but before the MOT slideshow is displayed.

AlbrechtL, 2019-12-06:
OK. I think I reproduced the issue by adding a `usleep(1e6);` just before the `return 0;` in `main.cpp`.

Commit https://github.com/AlbrechtL/welle.io/commit/d76a3b310a895c731f2ce3db94b800b5253cf9c2 should fix it. Please test it in your environment.

tenzap, 2019-12-06:
It looked promising, but I still could make it crash (2.1 + commit d76a3b3), unfortunately. :/
Error still looks the same at the same position `CRadioController::onNewImpulseResponse` that time.

[welle-io-20191206-210304.kcrash.txt](https://github.com/AlbrechtL/welle.io/files/3933808/welle-io-20191206-210304.kcrash.txt)

AlbrechtL, 2019-12-06:
Damn it! 
Can you try to stop the device before closing the application? Maybe this workarounds the race condition.

tenzap, 2019-12-06:
Stop the device ? How ? I haven't found any stop button yet.

AlbrechtL, 2019-12-07:
I meant this https://github.com/AlbrechtL/welle.io/commit/37d6b10543fb4db35797a7df931af5f0ccc42ef8.

Please try it again.

tenzap, 2019-12-08:
After many trials, I couldn't get welle.io to crash on closure (2.1+ commit d76a3b3 + commit 37d6b10). Seems fixed. Thanks a lot !

closed 2019-12-08 by AlbrechtL