AlbrechtL/welle.io #423: At startup, the device parameters from config file are not set

tenzap, 2019-09-26:
This is for SoapySDR, but maybe also for RTLSDR and Airspy

For soapy the parameters are : Antenna, clocksource, driverargs
for rtlsdr & airspy they are : biastee

So when starting welle.io with soapysdr with driverargs set, they are not used.

Moreover, there is kind of a mix up of things in SoapySDRSettings.qml. The value is filled with the wrong one...
        guiHelper.set**DriverArgs**SoapySdr(**clockSource**.text)
        guiHelper.set**ClockSource**SoapySdr(**driverArgs**.text)

tenzap, 2019-09-29:
A fix for SoapySDR is in PR #428
It seems it's the only device which may requires parameters to be initialized well. At least with version 0.6 of soapysdr

AlbrechtL, 2019-10-02:
Thanks for the improvement! We will add your pull request in version 2.1

tenzap, 2019-10-02:
I think this one should also go into 2.0. SopaySDR is just unusable for me in its current state in beta3.
I understand it is not just a simple change in the code, but I think this is necessary. 
If you could suggest me another way to avoid duplication of `enum class CDeviceID` I will do it asap.

SoapySDR scans all available devices and takes the first one which is "audio" in my case. Obviously this cannot work. Moreover the driver arguments are taken from the wrong place in the GUI, and even if I place them in the correct place, they are not considered by Welle.io at startup. 

So there is no way to configure soapy's driver arguments in the current state of 2.0beta3

AlbrechtL, 2019-10-03:
In this case if would be better to remove Soapy from the version 2.0. From another reason SoapySDR is not working in the AppImage (https://github.com/AlbrechtL/welle.io/issues/349) anyway.

I think in version 2.1 we can fix it and we have to find a way how to distrubute welle.io with SoapySDR.

tenzap, 2019-10-03:
The bug I describe is with SoapySDR 0.6, but with version 0.7 there is only one device reported, my rtl-sdr device which then works fine with welle.io

So :
- welle + Soapy0.6 -> KO (but OK with the fix)
- welle + Soapy0.7 -> OK

AlbrechtL, 2019-10-19:
I will release 2.0 soon (without any changes at the Soapy input). Then we can start working at version 2.1.

AlbrechtL, 2019-10-19:
For my point of view it would be nice to have the complete SoapySDR binaries included into the AppImage. If so we can use the latest Soapy version.

tenzap, 2019-10-19:
Usually on linux, the apps use the version of the libraries shipped with the distro. Debian stable ships with 0.6... So there should be a fix for those stucked on 0.6

mpbraendli, 2019-10-21:
About SoapySDR in the AppImage not using system device drivers, there was a previous discussion in https://github.com/AlbrechtL/welle.io/issues/349#issuecomment-465943600

I don't see a way to include SoapySDR inside the AppImage that works with system drivers (because device choice is up to the user, not us) on all distributions (version choice depends on distribution).

But I'd love to be proven wrong on that one :-)

AlbrechtL, 2019-10-22:
Oh I expected that the user can tell Soapy which driver to use. My idea was just put the Soapy libraries including the libraries for the common SDR devices like Hack RF, Lime, SDR etc. to the AppImage.
Why this should be not possible?

BTW: I never used Soapy so please correct me if I'm totally wrong.

mpbraendli, 2019-10-28:
It's not impossible, but it means we decide what devices the user may use, which goes against the purpose of SoapySDR.

However, we could stop saying "we support SoapySDR" and instead say "we support HackRF, LimeSDR and Ettus USRP through a bundled SoapySDR" so that it's clear that the user cannot add another device when he uses the AppImage.

If we do that, do we have to bundle all SoapySDR device drivers' dependencies? That would be libhackrf, uhd, LimeSuite. That'll increase the size of the AppImage. I do not know the idea and technicalities behind the AppImage well enough to know where we have to stop bundling things. I believe it would be better to have distributions carry recent versions in their repositories (debian has it in testing: https://packages.debian.org/bullseye/welle.io)

AlbrechtL, 2019-12-06:
I just realized that the Linux kernel has a native SDR API. https://www.kernel.org/doc/html/v4.12/media/uapi/v4l/dev-sdr.html
Does it make sense to support this API in order to get also other SDR devices working?

AlbrechtL, 2020-01-15:
@tenzap Can we close this issue?

tenzap, 2020-01-15:
Yes we can. Parameters for SoapySDR are now well restored/used :)

closed 2020-01-22 by AlbrechtL