AlbrechtL/welle.io #429: NewFeature: Add ability to set the qtquick style in the gui

tenzap, 2019-09-30:
I made a few changes to be able to select QtQuick Style at startup.
ComboBox displays all styles available on the system with 
- Default as 1st item, and
- System Auto as 2nd item. System Auto works as if the Style was not hardcoded.

Welle.io restart is needed so that the change is effective

[image]

I can make a PR if you would like to add that feature

AlbrechtL, 2019-10-02:
Wow! That is really great. Some people already requested it!
I can add it in version 2.1 if you send a pull request.

tenzap, 2019-10-04:
In PR #435

AlbrechtL, 2019-10-22:
#435 is merged!

closed 2019-10-22 by AlbrechtL