AlbrechtL/welle.io #432: Accessibility issues in welle.io web interface

domasofan, 2019-10-02:
Hi all,

I just tested welle.io version 2.0 Beta 3 on Windows 10 1903 64 Bit using the NVDA screen reader (daily builds).
The desktop GUI is inaccessible unfortunately though QT 5 should have some accessibility improvements and features.
Though the web interface looks pretty promising.
So i started welle-cli.exe and the web server.
I seem to have found some stations over here by providing the -c option on welle-cli to switch to the local channel for DAB+.
Though when you press play on one station there is no audio playing.
Have I forgotten something?
I guess it has something to do with the fast refreshing interface so the screen readers focus gets lost.
Screenreaders use a kind of buffer mode (browse mode) where the web page gets converted into a flat style design so you can read and navigate it better.
It can cope with pages that update but sometimes can loose focus on very fast updates.
Maybe there could be a option to turn off updates or something.

There is also focus mode or forms mode which allows you to use the page as sighted people interact with it. this is used in forms thats why it was called forms mode for a long time.

I also saw some accessible html5 tables to navigate using the keyboard. so you can have the screenreader stay in focus mode and not using browse mode.

The server also crashes when you switch the channel list to fast.
Its a html dropdown list i guess with a OnChange or OnSelect event which triggers when you use the screen readers forms mode.
That seems to make the server a little bit concerned when doing that too fast so it crashes.
Though you can go around with that by hitting alt+down arrow to open the list and then select and then close the list with alt+up arrow.
though the server seems to crash anyway.

That's all for the moment.

Greetings,
Simon

AlbrechtL, 2019-10-03:
@domasofan From my point of view we have to make welle.io accessible for screen readers. I personally don't have any experince in that. welle.io is based on Qt, so if you want to help us you can find some websites how to write a accessible Qt application.

domasofan, 2019-10-03:
Hi Albrecht and all, 

This is what i have found by just googling a bit: 

QT 5.13 Accessibility: 
https://doc.qt.io/qt-5/accessible.html

QT 5.13 accessibility classes: 
https://doc.qt.io/qt-5/accessibility.html

QT 5.13: Accessibility for QWidget Applications 
https://doc.qt.io/qt-5/accessible-qwidget.html

Accessible QML Type | Qt Quick 5.13.1 - Qt Documentation 
https://doc.qt.io/qt-5/qml-qtquick-accessible.html

Qt Blog | Accessibility: 
https://www.qt.io/blog/tag/accessibility

Accessibility for Qt Quick Applications | Qt 5.13: 
https://doc.qt.io/qt-5/accessible-qtquick.html

Maybe that helps. 

Greetings, 
Simon

AlbrechtL, 2019-10-19:
Thanks for helping! We put it to our task list.

tenzap, 2020-05-05:
@domasofan accessiblity support has been added. Can you please test with latest nightly and report back ?

domasofan, 2020-05-05:
Hi @tenzap,

Just tried version 20200430 b1589bd.
Though I seem not to notice much difference than  a few months ago.

Tested the web and the qt interface.
Have I forgotten anything?
There seem to be some channels in the qt interface though the screenreader seems to see many unlabeled buttons and a dropdown box where you can select all channels or favorites.

when i click or press enter on one of the unlabeled buttons nothing happens.

Greetings,
Simon

tenzap, 2020-05-05:
The change only concerns the GUI. It was implemented using the doc you referenced.

Your result is disappointing. I checked that on windows with NVDA and all labels were spoken out.
What do you actually expect ? Can you give a precise example on what is the expected behavior ?

domasofan, 2020-05-05:
Hi,

I am sure we will get there sometime.

I just tested with NVDA 2019.3.1 Stable and the 2 daily builds alpha 20086 and 20145 which is at the moment the newest one from this day.
Unfortunately I seem to get the same result.
When tabbing you just get a lot of buttons with no labels so NVDA reports Button, button, button, etc.
The controls are all accessible ones but not labeled as it seems.
Though nvda can detect the station names when using the object navigation which is a kind of mouse emulation.

Maybe I try to get someone who has more knowledge than me and can look into that issue.
Maybe even someone who understands qt and its accessibility stack.

We are not lost. Not yet.

Greetings,
Simon

AlbrechtL, 2020-07-12:
@domasofan We improved the accessibility in https://github.com/AlbrechtL/welle.io/pull/535 (Big thanks to @tenzap)

Can you test the latest version from https://bintray.com/albrechtl/welle.io/welle.io_nightly#files?

domasofan, 2020-07-24:
Hi @AlbrechtL,

Sorry for writing in pretty late.
Was on holiday.
I tested build 20200711.
Thanks the software is a little bit more accessible.
I have to use my mouse emulation to click on the station names.
There are a lot of unlabeled buttons on the main screen.
I guess that is the station list, because the names of the stations are next to the buttons.
Is it possible to label these buttons as well?
So that the station names are also on the buttons.
I also seem not to see all the stations that are nearby.
Swiss stations which i normally get pretty well seem not to be on the list.

On another note:
Maybe it would be a interesting idea to use a list box instead of buttons to select the station?

Greetings,
Simon

AlbrechtL, 2020-09-03:
Thanks for your feedback. So still a few things to do.

closed 2024-10-08 by AlbrechtL