AlbrechtL/welle.io #433: song saving of mp3 with title/interpreter

alexmyczko, 2019-10-03:
I know there has been closed issues already, but streamripper would just not get the song titles, and having the mp3 streams is great via https://github.com/AlbrechtL/welle.io/wiki/Using-the-welle-cli-Web-Interface

What would I do to get a ton of mp3 with the right start/stop and songname/interpreter (DLS) from the welle-cli web interface?

the web interface is jumpy (horiziontally as well as vertically), can that be improved?

andimik, 2019-10-03:
I already have improved the horizontal jumping by setting a fixed width for the Dynamic Label column. So you could edit the file yourself as well.

alexmyczko, 2019-10-04:
How did you? I did like this:

```
 table, th, td {
     border: 1px solid black;
     border-collapse: collapse;
+    width: 100%;
 }
 th, td {
     padding: 5px;
     text-align: left;
+    width: auto;
 }
 th#dls {
      width: 300px;
```

AlbrechtL, 2019-10-19:
What is your use case? I mean I totally understand it but do you do it for personal use or some professional projects?

alexmyczko, 2019-10-21:
have some music replayable, while offline, personal only

mpbraendli, 2019-10-21:
Don't use the web page for such tasks, get the `mux.json` directly and parse it externally with a script.

AlbrechtL, 2019-10-22:
@alexmyczko Is your question answered?

mpbraendli, 2019-10-28:
(I interpreted @alexmyczko's thumbs-up as "yes")

closed 2019-10-28 by mpbraendli