AlbrechtL/welle.io #436: [macos macports] - rtl-sdr issues

tenzap, 2019-10-12:
This is on macos El Capitan 10.11.6

I installed the various libraries and also the qt libraries with macports.
Then I compiled welle.io with cmake (with both rtlsdr & soapysdr)

At startup, as soon as I start playing a station, or start a scan, welle.io blocks with a "occupied" mouse cursor.
This is when launching welle this way (almost every time)
`./welle-io`

The log is as bellow
[log.txt](https://github.com/AlbrechtL/welle.io/files/3720972/log.txt)

However, I noticed that if I run it with this command, most time the issue doesn't appear...
`./welle-io 2> /dev/null > /dev/null`

On another hand, when using the rtl-sdr plugin, when browsing in firefox while welle.io is playing in the background it seems synchronisation is lost at every action I do in firefox. The thing is that at every click in firefox, playing is interrupted for a short time until it somehow synchronises again.

There is never such a problem when choosing soapysdr (with SoapyRtlSdr module) whatever the command above

AlbrechtL, 2019-10-19:
I don't own a Mac so I don't have any experience with it.

For the mouse behavior I don't have a explanation. Did you enable the console view inside the expert mode?

The behavior for sync lost during web browsing I know from low powered CPU like Intel Atom (2015) and Raspberry Pi 2/3.

tenzap, 2019-10-19:
expert view was not enabled.
Well, the mouse icon means busy, and welle-io  has to be killed manually. That's the only way to get out of it

AlbrechtL, 2019-10-22:
Which CPU has your Mac?

tenzap, 2019-10-23:
I don't own a mac, but that mac is 
Processor Speed: | 2.66 GHz | Processor Type: | Core 2 Duo (E8135)

But I don't think the freeze is CPU linked since 
- there is no freeze with Soapy + soapyrtlsdr 
- but it freezes with only rtl-sdr

AlbrechtL, 2019-10-25:
According to https://www.cpubenchmark.net/cpu.php?cpu=Intel+Core2+Duo+E8135+%40+2.66GHz&id=949 the Intel E8135 CPU is 10 years old and has only two cores. Even the CPU mark of 1761 shows that this is the absolutely minimum to run welle.io. I don't know how macOS renders the graphics but if the rendering is done with the CPU then the CPU power is ways to low.

I'm pretty sure that issues comes with the low CPU power.

AlbrechtL, 2019-10-26:
Can you try the 2.0 beta 3 (https://github.com/AlbrechtL/welle.io/tree/v2.0-beta3)?

tenzap, 2019-10-26:
This was with 2.0beta3 until commit 801e483

tenzap, 2019-10-26:
Hmm... interesting... 
"pure" 2.0beta3 -> Works (both with & without autoplay)
"pure" 2.0beta3 + #431 + 'no autoplay' -> Works
"pure" 2.0beta3 + #431 + 'autoplay' -> Freezes at startup

So it seems #431 is in cause.

AlbrechtL, 2019-11-24:
Any news here?
@tenzap Do you have a fix?

tenzap, 2019-11-24:
should be fixed by #460 I guess

AlbrechtL, 2019-11-24:
>concerning the freeze should be fixed by #460

Great! Then I can close this issue again ;-)

> But concerning the sync loss, I have no clue

The sync lost is coming from the weak CPU power.

closed 2019-11-24 by AlbrechtL