AlbrechtL/welle.io #440: [MacOS] warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12

tenzap, 2019-10-18:
Compilation on macOS >= 10.12 gives this warning :

`warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]`

```
/opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_multimedia_welle.io/welle.io/work/welle.io-801e483ce5340cb816cff63743706c245d8c5fe6/src/various/ringbuffer.h:213:13: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]
/opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_multimedia_welle.io/welle.io/work/welle.io-801e483ce5340cb816cff63743706c245d8c5fe6/src/various/ringbuffer.h:222:13: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]
/opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_multimedia_welle.io/welle.io/work/welle.io-801e483ce5340cb816cff63743706c245d8c5fe6/src/various/ringbuffer.h:259:17: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]
/opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_multimedia_welle.io/welle.io/work/welle.io-801e483ce5340cb816cff63743706c245d8c5fe6/src/various/ringbuffer.h:297:17: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]
/opt/local/var/macports/build/_opt_bblocal_var_buildworker_ports_build_ports_multimedia_welle.io/welle.io/work/welle.io-801e483ce5340cb816cff63743706c245d8c5fe6/src/various/ringbuffer.h:349:13: warning: 'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use std::atomic_thread_fence() from <atomic> instead [-Wdeprecated-declarations]
```

mpbraendli, 2019-10-19:
That looks like an opportunity to replace the big block of macro definitions inside `src/various/ringbuffer.h` by something simpler.

Actually, I would even prefer if we could replace the whole ringbuffer implementation by something more understandable than the current lockfree code... But maybe using the `std::atomic_thread_fence()` is good enough. @tenzap do you want to try?

tenzap, 2019-10-19:
No sorry I'm not a real dev and all the thread things is actually too complicated for me

AlbrechtL, 2019-10-19:
Duplicated, please see https://github.com/AlbrechtL/welle.io/issues/336.

closed 2019-10-19 by AlbrechtL

mpbraendli, 2019-10-21:
> No sorry I'm not a real dev and all the thread things is actually too complicated for me

Well, seeing all your pull requests and your recent involvement I wouldn't dare to say that!

If you want to check if the idea works, try:

- adding `#include <atomic>`
- replacing the block of macros with

```
#define PaUtil_FullMemoryBarrier() std::atomic_thread_fence()
#define PaUtil_ReadMemoryBarrier() std::atomic_thread_fence()
#define PaUtil_WriteMemoryBarrier() std::atomic_thread_fence()
```

and let us know if you can get it working, and let us know with a comment in issue #336